zephyrproject-rtos/zephyr #41017: Defining a new USB string descriptor breaks the stack

abxy, 2021-12-08:
**Describe the bug**
While trying to implement a USB video device I encountered
unexpected errors in `dmesg` when plugging in the device.

I managed to to track the problem to the single string descriptor I had
defined. I believe my string descriptor was placed before the other string descriptors
(`string_descr` in `usb_descriptor.c`) by the linker, causing the `usb_fix_descriptor`
routine to break in ways I don't yet understand.
I have created a PR https://github.com/zephyrproject-rtos/zephyr/pull/41014  that fixes my issue by disambiguating the order
between string descriptor zero and other string descriptors.

**To Reproduce**
Since the issue is with under specified section ordering I have not
been able to find a way to reproduce this with any in-tree boards or samples.

I'm working on my USB video class driver out-of-tree application,
that might affect the link order.

The following config options were helpful in discovering the cause:
```
CONFIG_USB_DEVICE_LOG_LEVEL_DBG=y
CONFIG_LOG_BUFFER_SIZE=4096
```

**Logs and console output**
Log output during descriptor fixup:
```
[00:00:00.000,000] <dbg> usb_descriptor.usb_get_device_descriptor: __usb_descriptor_start 0x20000190
[00:00:00.000,000] <dbg> usb_descriptor.usb_get_device_descriptor: __usb_descriptor_end 0x20000298
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: Configuration descriptor 0x200001a2
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: Association descriptor 0x200001ab
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: Interface descriptor 0x200001b3
[00:00:00.000,000] <dbg> usb_video.video_interface_config: function_string_desc: 0x2000024a
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: Interface descriptor 0x200001e4
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: Interface descriptor 0x20000234
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: Skip alternate interface
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: Interface descriptor 0x2000023a
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: Skip alternate interface
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: Endpoint descriptor 0x20000243
[00:00:00.000,000] <dbg> usb_descriptor.usb_validate_ep_cfg_data: Fixing EP address 80 -> 81
[00:00:00.000,000] <dbg> usb_descriptor.usb_validate_ep_cfg_data: endpoint 0x81
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: Now the wTotalLength is 168
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: idx_max 1, ascii_idx_max 0, buf 0x20000254
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: char 	: 9, idx 0 -> 1
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: idx_max 11, ascii_idx_max 5, buf 0x20000258
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: char R : 52, idx 5 -> 11
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: char Y : 59, idx 4 -> 9
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: char H : 48, idx 3 -> 7
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: char P : 50, idx 2 -> 5
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: char E : 45, idx 1 -> 3
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: char Z : 5a, idx 0 -> 1
[00:00:00.000,000] <dbg> usb_descriptor: Serial Number
                                         30 39 51 12 00 31 80 0e                          |09Q..1..         
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: idx_max 13, ascii_idx_max 6, buf 0x20000266
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: char 1 : 31, idx 6 -> 13
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: char 1 : 31, idx 5 -> 11
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: char 5 : 35, idx 4 -> 9
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: char 9 : 39, idx 3 -> 7
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: char 3 : 33, idx 2 -> 5
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: char 0 : 30, idx 1 -> 3
[00:00:00.000,000] <dbg> usb_descriptor.ascii7_to_utf16le: char 3 : 33, idx 0 -> 1
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x2000032d
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x2000032f
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x20000331
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x20000333
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x20000335
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x20000337
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x20000339
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x2000033b
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x2000033d
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x2000033f
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x20000341
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x20000343
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x20000345
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x20000347
[00:00:00.000,000] <dbg> usb_descriptor.usb_fix_descriptor: try to fix next descriptor at 0x20000349
[00:00:00.000,000] <err> usb_descriptor: Incomplete device descriptor
[00:00:00.000,000] <err> usb_descriptor: Failed to fixup USB descriptor
[00:00:00.000,000] <err> usb_device: Failed to configure USB device stack
```

Output from dmesg when plugging in device with broken descriptor.
```
[36837.798905] usb 1-9.3: new full-speed USB device number 60 using xhci_hcd
[36837.799222] usb 1-9.3: Device not responding to setup address.
[36838.007260] usb 1-9.3: Device not responding to setup address.
[36838.214879] usb 1-9.3: device not accepting address 60, error -71
[36838.398888] usb 1-9.3: new full-speed USB device number 61 using xhci_hcd
[36838.399201] usb 1-9.3: Device not responding to setup address.
[36838.607032] usb 1-9.3: Device not responding to setup address.
[36838.818905] usb 1-9.3: device not accepting address 61, error -71
[36838.819032] usb 1-9-port3: unable to enumerate USB device
```

**Environment:**
 - OS: Debian 11 (5.10.0-9-amd64)
 - Toolchain: Zephyr SDK 0.13.1
 - Zephyr tag v2.7.0

abxy, 2021-12-08:
I've convinced myself that the corruption happens when another string descriptor is placed before the string descriptors in `struct usb_string_desription string_descr` and their indices are shifted so `usb_fix_ascii_sn_string_descriptor` is called with the product string descriptor instead of the serial number descriptor.

In the default config the product string is smaller than the serial number, so `memcpy` writes out of bounds, overwriting the `bLength` field of the next descriptor, corrupting the descriptor structure.

abxy, 2021-12-09:
I found confirmation in `zephyr.map`. In the snippet below you can see that the string descriptor (identifiable by the section ` .usb.descriptor_primary.4`) that I defined in `usb_video.c` is placed before the string descriptors defined by `usb_descriptor.c`.

```
 *(SORT_BY_ALIGNMENT(.usb.descriptor))
 *(SORT_BY_NAME(SORT_BY_ALIGNMENT(.usb.descriptor*)))
 .usb.descriptor_primary.0
                0x0000000020000190       0x1b zephyr/libzephyr.a(usb_descriptor.c.obj)
 .usb.descriptor_primary.1.0
                0x00000000200001ab       0x9f app/libapp.a(usb_video.c.obj)
 .usb.descriptor_primary.4
                0x000000002000024a        0x8 app/libapp.a(usb_video.c.obj)
 .usb.descriptor_primary.4
                0x0000000020000252       0x44 zephyr/libzephyr.a(usb_descriptor.c.obj)
 .usb.descriptor_primary.5
                0x0000000020000296        0x2 zephyr/libzephyr.a(usb_descriptor.c.obj)
                0x0000000020000298                __usb_descriptor_end = .
```

jfischer-no, 2021-12-09:
For clarification, string descriptors are optional. There is no straightforward support in the current USB device support for the string descriptors defined in class or application code. Event with #41014, there is no solution to avoid indexes conflict as they are not part of string descriptor and we do not have a container for that.

abxy, 2021-12-09:
According to the USB Video Class Specification there is one mandatory string descriptor. The rest are optional as you say.

Can you clarify what you mean about index conflicts? I don't currently have a need to explicitly control the indexes. From my (very) limited understanding of the USB protocol there is no need to control the string descriptor ordering (aside from index 0). You just have to make sure to point to the right string from your other descriptors and it seems like `usb_get_str_descriptor_idx` can be used for that purpose.

PR #41014 (2022-02-03): usb: Add macro for user defined string descriptors
The new macro USBD_STRING_DESCR_USER_DEFINE works like
USBD_STRING_DESCR_DEFINE with the exception of being
ordered strictly after it. The new macro is needed to
ensures that user defined string descriptors can be added
without disturbing the order of string descriptors
defined by the usb subsystem.

Fixes  #41017
closed 2022-02-03 by carlescufi