zephyrproject-rtos/zephyr #41039: Xtensa High priority interrupt interrupt excption interrupt causes double excption

hongshui3000, 2021-12-09:
**Describe the bug**
The zephyr xtensa architecture does not seem to take this situation into consideration:
For example, the excption interrupt level is less than the highest priority interrupt. When the high priority interrupt preempts the excption interrupt (such as WindowOverflow8), the entire stack will be destroyed
[image]

andyross, 2022-01-20:
Looking at this for the first time.  I'm sorry, I don't understand what you want Zephyr to do here?  We took an exception, it got preempted by an interrupt that Zephyr doesn't manage and landed in a handler with code written by the application.

How does Zephyr prevent the application code from breaking Zephyr state?  You need to write your custom interrupt handlers to be safe when they preempt important things like window exceptions.

I understand that's extremely difficult.  During a window exception you don't have (and probably can't recover reliably) a stack pointer.  But it's not impossible.  Grab a SR that isn't in use (we have a brand new feature to help you with this) and do a musical chairs entry to carefully save off the GPR state.  And needless to say the resulting exception code needs to be built to the CALL0 ABI, or be written entirely in assembly.

But it's not Zephyr's problem to solve.  The interrupts we emit code for work reliably.

hongshui3000, 2022-01-21:
I can understand what you're saying,
1. the issue is that people porting xtensa arch or people using xtensa may encounter this problem.
2. This should be classified as a problem with xtensa arch. If the code implementing high priority interrupts will need to call the underlying code of the current architecture (such as thread switching, etc.), if every ordinary user using high priority calls the underlying code It will be risky.
3. The interrupt registration function "IRQ_CONNECT(DT_INST_IRQN(0), DT_INST_IRQ(0, priority), i2c_my_isr, DEVICE_DT_INST_GET(0), 0)" in zephyrrtos will register any priority interrupt to the current interrupt code, and currently The code is actually unable to support high priority interrupts. If the user does not notice, it will cause rtos to crash, and it will crash randomly (interruption is random), which is difficult for ordinary people to solve

I think currently zephyrrtos should at least deny xtensa high priority interrupt registration or alert the user, or improve xtensa arch to meet high priority interrupt registration

andyross, 2022-01-25:
Agreed, let's mark this as an enhancement instead of a bug.  Zephyr does not currently support applications registering interrupts that interrupt exception handlers.  Applications that do so need to be coded very carefully.  It would be good if Zephyr had an answer here, but it doesn't.  We should add that as a feature.

hongshui3000, 2022-01-26:
> 

agree completely