zephyrproject-rtos/zephyr #41043: Sporadic Bus Fault when using I2C on a nrf52840

caco3, 2021-12-09:
We have an application based on a nrf52840 and zephyr 2.7.0.
The system is used for a measurement device. We run an external ADC at 10 kSamples/s and read its data through SPI DMA transfers. After 30 samples (=> 333 Hz) we trigger an Interrupt Handler which copies the data into another buffer. The SPI and the Timer are connected via PPI.

Also we have several I2C devices with which we need to communicate every 10s. For this we use the blocking `i2c_read` and `i2c_write` functions.

Both parts (Measurement Data transfer via SPI DMA as well as the I2C communications) work stable forever if let run by themselves!
How ever when we let them run in parallel, we see occational BUS Faults resp. CPU Lockups! Sometimes after minutes, sometimes after 1h.

Most of the time the system just restarts and the CPU Lockout flag is set. Sometimes we also see an error:
```
[..]
[00:10:03.643,249] <err> os: ***** BUS FAULT *****
[00:10:03.643,280] <err>*** Booting Zephyr OS build zephyr-v2.7.0  ***
[..]
[00:00:00.254,272] <inf> Reset Reason: CPU-Lockup (0x00000008)
[..]
```

Part of the `prj.conf`:
```
CONFIG_EXCEPTION_STACK_TRACE=y

CONFIG_I2C=y
CONFIG_I2C_LOG_LEVEL_DBG=y

CONFIG_NRFX_GPIOTE=y
CONFIG_SPI=y
CONFIG_SPI_NRFX=y
CONFIG_NRFX_PPI=y
CONFIG_NRFX_SPIM=y
CONFIG_NRFX_TIMER=y
CONFIG_NRFX_TWI=y
CONFIG_NRFX_TWI0=y
CONFIG_NRFX_TWI1=y
```

Questions:
- [ ] Is there a way to get more information on the crashs. Eg. is there a config flag to get more printouts on the fault?
- [ ] I am unsure if the system uses the NRFX drivers or if Zephyr uses another generic driver for I2C

DariusBabrauskas, 2021-12-10:
Hi. Are you disable compilation optimization?
If yes, then possible stack overflow ().  Try increase stack size. Example.:
CONFIG_MAIN_STACK_SIZE=2048

Maybe this configs can help you. (zephyr 2.5)
#CONFIG_SIZE_OPTIMIZATIONS=y
CONFIG_NO_OPTIMIZATIONS=y
#enable stack overflow test
CONFIG_STACK_SENTINEL=y
CONFIG_INIT_STACKS=y
CONFIG_STACK_USAGE=y
CONFIG_REBOOT=y

CONFIG_DEBUG_COREDUMP=y
CONFIG_DEBUG_COREDUMP_MEMORY_DUMP_MIN=y
CONFIG_DEBUG_COREDUMP_BACKEND_LOGGING=y
CONFIG_DEBUG=y
CONFIG_DEBUG_INFO=y

caco3, 2021-12-10:
Dear @DariusBabrauskas 

Thanks for your input!
Stack Overflowing is a good hint.  I increased now all my stacks of my threads by factor 4..10 and also the following ones:
```
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=6000
CONFIG_BT_RX_STACK_SIZE=6000
CONFIG_MAIN_STACK_SIZE=8096
CONFIG_IDLE_STACK_SIZE=8096
CONFIG_ISR_STACK_SIZE=8096
```
but the issue still occurs, so this seems not to be the issue.

I also added all of the following flags, but still do not get a proper trace on the fault:
```
CONFIG_DEBUG_COREDUMP=y
CONFIG_DEBUG_COREDUMP_MEMORY_DUMP_MIN=y
CONFIG_DEBUG_COREDUMP_BACKEND_LOGGING=y
CONFIG_DEBUG=y
CONFIG_DEBUG_INFO=y
```

`CONFIG_NO_OPTIMIZATIONS` or `CONFIG_INIT_STACKS` seems not to help either, althrough i now saw once a bit more:
```
[00:05:05.038,665] <err> os: ***** BUS FAULT *****
[00:05:05.038,665] <err> os:   Instruction bus▒*** Booting Zephyr OS build zephyr-v2.7.0  ***
```
`CONFIG_STACK_SENTINEL` I  had already.

`CONFIG_STACK_USAGE` does not work for me, I get a compilation error:
```
FAILED: zephyr/CMakeFiles/zephyr.dir/soc/arm/nordic_nrf/common/soc_nrf_common.S.obj 
ccache /opt/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.7.0 -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/opt/zephyrproject-v2.7.0/zephyr/kernel/include -I/opt/zephyrproject-v2.7.0/zephyr/arch/arm/include -I/opt/zephyrproject-v2.7.0/zephyr/include -I/home/wrcd/rcd-sensor_build/zephyr/include/generated -I/opt/zephyrproject-v2.7.0/zephyr/soc/arm/nordic_nrf/nrf52 -I/opt/zephyrproject-v2.7.0/zephyr/lib/libc/newlib/include -I/opt/zephyrproject-v2.7.0/zephyr/subsys/bluetooth -I/opt/zephyrproject-v2.7.0/zephyr/subsys/settings/include -I/opt/zephyrproject-v2.7.0/modules/hal/cmsis/CMSIS/Core/Include -I/opt/zephyrproject-v2.7.0/modules/hal/nordic/nrfx -I/opt/zephyrproject-v2.7.0/modules/hal/nordic/nrfx/drivers/include -I/opt/zephyrproject-v2.7.0/modules/hal/nordic/nrfx/mdk -I/opt/zephyrproject-v2.7.0/zephyr/modules/hal_nordic/nrfx/. -I/opt/zephyrproject-v2.7.0/modules/lib/tinycbor/zephyr/../include -I/opt/zephyrproject-v2.7.0/modules/crypto/tinycrypt/lib/include -fstack-protector-all -O2 -imacros /home/wrcd/rcd-sensor_build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -xassembler-with-cpp -imacros /opt/zephyrproject-v2.7.0/zephyr/include/toolchain/zephyr_stdint.h -D_ASMLANGUAGE -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -opt3 -fno-reorder-functions -fno-defer-pop -fstack-usage -fmacro-prefix-map=/home/wrcd/rcd-sensor=CMAKE_SOURCE_DIR -fmacro-prefix-map=/opt/zephyrproject-v2.7.0/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/opt/zephyrproject-v2.7.0=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=nano.specs -MD -MT zephyr/CMakeFiles/zephyr.dir/soc/arm/nordic_nrf/common/soc_nrf_common.S.obj -MF zephyr/CMakeFiles/zephyr.dir/soc/arm/nordic_nrf/common/soc_nrf_common.S.obj.d -o zephyr/CMakeFiles/zephyr.dir/soc/arm/nordic_nrf/common/soc_nrf_common.S.obj -c /opt/zephyrproject-v2.7.0/zephyr/soc/arm/nordic_nrf/common/soc_nrf_common.S
cc1: error: too many filenames given; type 'cc1 --help' for usage
cc1: fatal error: zephyr/CMakeFiles/zephyr.dir/soc/arm/nordic_nrf/common/soc_nrf_common.S.d: No such file or directory
compilation terminated.
```
Does this work for you on Zephyr 2.7.0?

caco3, 2021-12-10:
my bad, `CONFIG_NO_OPTIMIZATIONS` works now and gives me a trace: 
```
[00:00:00.450,317] <err> os: ***** MPU FAULT *****
[00:00:00.450,347] <err> os:   Stacking error (context area might be not valid)
[00:00:00.450,378] <err> os:   Data Access Violation
[00:00:00.450,378] <err> os:   MMFAR Address: 0x2001ff18
[00:00:00.450,408] <err> os: r0/a1:  0x00000000  r1/a2:  0x0002cc89  r2/a3:  0xaaaaaaaa
[00:00:00.450,439] <err> os: r3/a4:  0xa780c700 r12/ip:  0x00000000 r14/lr:  0xaaaaaaaa
[00:00:00.450,469] <err> os:  xpsr:  0x33483c00
[00:00:00.450,469] <err> os: Faulting instruction address (r15/pc): 0x6af033ab
[00:00:00.450,500] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:00.450,531] <err> os: Current thread: 0x20013600 (unknown)
```

(I had it overwritten further down in `proj.conf`)

DariusBabrauskas, 2021-12-10:
>my bad, CONFIG_NO_OPTIMIZATIONS works now and gives me a trace:
It not bad. But  if you enabled CONFIG_NO_OPTIMIZATIONS=y, then often occurs "Stack overflow". 

> [00:00:00.450,500] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
> [00:00:00.450,531] <err> os: Current thread: 0x20013600 (unknown)
What name of thread "unknown"?  Set  breakpoint  at point printing  "Current thread:"  and see which thread,then increase it size.

caco3, 2021-12-15:
Thanks for your input!
I traced it further down and learned that the crashes are caused by classical Buffer Overflows.

I will close this issue as the root cause is not related to I2C at all.

Just some notes for other readers:
- `CONFIG_NO_OPTIMIZATIONS` mostly didn't work for me, I got immediate Stack overflow.
- I get crashs because a function pointer gets altered and points to an invalid code fragment. The crahss seem to be such immanent that I do not get a stack trace in such case as shown in the first post (no crash information logged).
- `CONFIG_STACK_USAGE ` was not really useful for me, the suggested tool (https://github.com/zephyrproject-rtos/zephyr/issues/2566#issuecomment-601780283) didn't give me the expected information.

closed 2021-12-15 by caco3