zephyrproject-rtos/zephyr #41047: boards: pinetime: display backlight via GPIO

caspermeijn, 2021-12-09:
**Is your enhancement proposal related to a problem? Please describe.**
With the main Zephyr samples, the backlight of the [Pinetime](https://docs.zephyrproject.org/latest/boards/arm/pinetime_devkit0/doc/index.html) board is not enabled, therefore the picture of the display is not visible.

The backlight is enabled via GPIO. There are three GPIO, each with a different resistor, for three brightness levels. But due to the resistor this is not a very efficient way to reduce the brightness. It seems the best way is to use PWM on the pin for highest brightness.

There is currently no way to connect a backlight to a display driver. Some display drivers are able to set the brightness of the display controller. Most drivers will just return `-ENOTSUP` when changing the brightness. There is no display driver that supports a GPIO pin as backlight.

**Describe the solution you'd like**
Add a [led-backlight](https://elixir.bootlin.com/linux/latest/source/Documentation/devicetree/bindings/leds/backlight/led-backlight.yaml) devicetree binding similar to Linux. Then add a reference to the `led-backlight` from the st7789v display driver. This way the hardware connection is described in the devicetree.

Then extend the display driver to initialize the GPIO by navigating the binding to using a series of `DT_` macros. In the `set_brightness` function the GPIO can be controlled (setting on when `brightness != 0`). This is not the best solution, but it is a quick method to enable the backlight.

**Describe alternatives you've considered**
1) Adding the GPIO binding directly to the display devicetree entry. This seems to be the easiest option, but it is not very extend-able.

2) Using the led drivers for activating the GPIO seems to be the better long term solution. This also makes it possible to enable PWM on the backlight. I would like to do that on a later enhancement. First make it work, then make it work better ;-)

**Additional context**

ck-telecom, 2021-12-10:
There is a led_pwm driver in drivers/led/led_pwm.c, whether we could using this driver in display driver, invoke the APIs like invoke SPI bus driver?

caspermeijn, 2021-12-10:
> There is a led_pwm driver in drivers/led/led_pwm.c, whether we could using this driver in display driver, invoke the APIs like invoke SPI bus driver?

Yes, using led_pwm driver is what I meant with alternative 2. I was not able to prototype this solution before, but now I can and it seems easy. I will change the description to reflect this new insight.

caspermeijn, 2021-12-10:
I know again what the problem is with using the led_pwm driver. If you have a DTS phandle to a single led, then you can't retreive the led number. The led number is basically the index of the led in the `gpio-leds` compatible.

I can't find an example of a DTS reference to a led in the main Zephyr repo. All led related examples loop over all leds in the first `gpio-leds` compatible.

@ck-telecom Do you know of such example?

ck-telecom, 2021-12-11:
@caspermeijn I think it shall looks like this:

	pwmleds {
		compatible = "pwm-leds";
		pwm_led0: pwm_led_0 {
			pwms = <&pwm0 13>;
		};
	};

&st7789v {
pwm-backlight = <&pwm_led0>;
}

caspermeijn, 2021-12-11:
@ck-telecom Yes, I also think it should be something like that. But how do you use that reference to set the brightness of the pwm-led.

I try to finish my proof of concept code and I can give you a proper example.

ck-telecom, 2021-12-11:
This way is a wapper of pwm-leds, you could invoke pwm-les APIs to control brightness, but you should binds the pwm-leds device, can you try to unterstand drivers/led/led-pwm.c, similar to invoke pwm APIs, but in st7789v driver, you should invoke pwm-leds APIs like set_brightness().

ck-telecom, 2021-12-11:
I have a look at Linux backlight, it use /sys/class/leds/<some node> to control backlight, the don't combo this in the linux framebuffer drivers, so I am not sure it is a good idea to control pwm based backlight in a display driver, maybe we shall control pwm backlight using pwm-leds driver, that is fine to most applications

ck-telecom, 2021-12-14:
DT_NODELABEL(pwn_led0) maybe helpful

caspermeijn, 2021-12-14:
@ck-telecom I finished my prove of concept. See TODO at the link below for the problem I had with the reference.