zephyrproject-rtos/zephyr #41052: tests-ci : portability: posix: fs.tls.newlib test Build failure

hakehuang, 2021-12-10:
**Describe the bug**
fs.tls.newlib test is Build failure on v2.7.99-2147-g183328a4fbdb on mimxrt685_evk_cm33

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt685_evk_cm33  --sub-test portability.posix
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**
```
None
    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: v2.7.99-2147-g183328a4fbdb

hakehuang, 2021-12-10:
dup with zephyr#41052

dleach02, 2022-01-11:
@hakehuang, does this error still occur? 

If so, can you provide a more focused target to reproduce this error? I run the twister command you specified and it ends up trying to build 543 test cases. After 257 of them built without error I stopped it because I'm running out of disk space.

hakehuang, 2022-01-12:
@dleach02  in latest build this is ok.

closed 2022-01-12 by hakehuang