zephyrproject-rtos/zephyr #41058: stm32h723 : application gets hung during spi_transceive() operation

Kshitij4kk, 2021-12-10:
**Describe the bug**
I am trying to interface ADXL345  accelerometer using SPI communication to my STM32H723ZG board. 
I am using the SPI1 device for this purpose.

I have done the following modifications in dts overlay file to enable the device node.
`
/**** pinctrl new additions ****/
&pinctrl {
	spi1_sck_pa5: spi1_sck_pa5 {
	    pinmux = < 0x505 >;
	    bias-pull-up;
	    slew-rate = "very-high-speed";
	};
	
	spi1_mosi_pd7: spi1_mosi_pd7 {
	    pinmux = < 0x3705 >;
	    bias-pull-down;
	    slew-rate = "very-high-speed";
	};
	
	spi1_miso_pa6: spi1_miso_pa6 {
		pinmux = < 0x605 >;
		bias-pull-down;
		slew-rate = "very-high-speed";
	};
         
};

/**** Enabling for accelerometer ***/
&spi1 {
	status = "okay";
};
`

I have used the following code for spi configuration:
`
	struct spi_config spi_cfg = {
		.operation = SPI_WORD_SET(8) | SPI_OP_MODE_MASTER |
			     SPI_MODE_CPOL | SPI_MODE_CPHA | SPI_TRANSFER_MSB,
		.frequency = 1074218U,
		.slave = 0,
		.cs = NULL
	};
`
Other than that I am using following code to read the device id of ADXL345 to ensure the communication channel is up and running
`
	/* Try to read ADXL345 device ID with above configuration to ensure communication channel is up and running */
	uint8_t tx_buf[1], rx_buf[1];
	tx_buf[0] = prepare_tx_data(READ_DEVID_REG, true, false);  /* This will just format the data in format accepted by ADXL345 */

	struct spi_buf spi_buf_tx[1], spi_buf_rx[1];
	spi_buf_tx[0].buf = &tx_buf[0];
	spi_buf_tx[0].len = 1;
	spi_buf_rx[0].buf = &rx_buf[0];
	spi_buf_rx[0].len = 1;

	struct spi_buf_set tx, rx;
	tx.buffers = &spi_buf_tx[0];
	tx.count = 1;
	rx.buffers = &spi_buf_rx[0];
	rx.count = 1;

	if (0 == spi_transceive(spi_dev, &spi_cfg, &tx, &rx)) {
		LOG_INF("SPI transceive operation successful for reading device ID!");
		LOG_INF("Read device ID: %x", rx_buf[0]);
		/* Actual Device Id: 0xE5  Expected Output: 0x1B (2's complement of 0xE5)*/
	} else {
		LOG_ERR("SPI transceive operation failed for reading device ID!");
		return ERR_SPI_OPERATION_FAILED;
	}
`

I can't see any response once the spi_transceive() routine is called. 

It would be great if I could get some guidance on this.

erwango, 2021-12-10:
You're likely hit by the issue that is trying to be addressed here: https://github.com/zephyrproject-rtos/zephyr/pull/40998

Kshitij4kk, 2021-12-13:
Thanks @erwango for the information.
I have not been able to perform any successful transaction on the SPI device yet so mostly I am stuck with the specified issue here.
So can we conclude that we can't use SPI1/2/3 currently on stm32h7 boards with Zephyr driver or is there some work-around possible?

erwango, 2021-12-13:
> So can we conclude that we can't use SPI1/2/3 currently on stm32h7 boards with Zephyr or is there some work-around possible?

Current situation is that there are some specific clock configuration cases where SPI baud rate caclulation goes wrong, cf https://github.com/heinwessels. Otherwise it is expected to be functional.

FRASTM, 2021-12-22:
I see similar lock with the spi_loopback tests with DMA but also in interrupt mode
@Kshitij4kk, I guess the SPI in your example is _not_ using DMA transfer

FRASTM, 2022-01-06:
@Kshitij4kk, Could you please check if PR https://github.com/zephyrproject-rtos/zephyr/pull/41560  helps

Kshitij4kk, 2022-01-11:
Thanks, I will be closing this issue.

closed 2022-01-11 by Kshitij4kk