zephyrproject-rtos/zephyr #41073: twister: no way to specify arguments for the binary zephyr.exe

apsigma, 2021-12-10:
**Is your enhancement proposal related to a problem? Please describe.**
When building/running tests on the native_posix platform, and the test has some BT HCI parts, the zephyr.exe complains that it doesn't have a bluetooth device, `Error: Bluetooth device missing. Specify one using --bt-dev=hciN`. I can then manually add the `--bt-dev=hci0` for example, and in my case the test then works.

When building/running with `twister`, there is no way for me to add extra args for the binary. The extra args field that is there today is only used for CMake.

Also, when using twister with this problem, it is very unclear that that is the real problem. Only by running the binary manually you can see that that is indeed the problem.

**Describe the solution you'd like**
I would like to have another argument to twister that allows me to pass arguments to the binary. I guess the best way to do this would be to specify it in the `testcase.yaml`, since then it will only be for that binary, and not for others that might be picked up by twister.

**Describe alternatives you've considered**
No real alternative identified.

**Additional context**
This has been tried on `<v2.6.0-ncs1-rc3>`, but I looked through `main`, and saw no obvious changes/additions to this part there.

tristan-google, 2022-11-18:
I see my issue was marked as a duplicate. Is there still interest in having such a feature? I have some time to work on this.

nashif, 2022-11-18:
> I see my issue was marked as a duplicate. Is there still interest in having such a feature? I have some time to work on this.

duplicate does not mean we do not want it, we should just track things in one place, and this one was earlier, so I closed yours. Please go ahead and submit a PR if you have anything ready.

tristan-google, 2022-11-30:
I uploaded a proposal: #52697

nashif, 2023-01-30:
proposal merged, closing.

closed 2023-01-30 by nashif