zephyrproject-rtos/zephyr #41074: can_mcan_send sends corrupted CAN frames with a byte-by-byte memcpy implementation

MJayat, 2021-12-10:
**Describe the bug**
Tested on board nucleo_g0b1re , it seems drivers/can/can_mcan.c has a problem.

If the implementation of memcpy compiled into Zephyr does a byte-by-byte copy, even on word-aligned addresses, then the tx_buffer in dedicated CAN Message RAM is corrupted. I think this is because it is supposed to be only accessed on word-aligned addresses.

The following copy of `struct can_mcan_tx_buffer_hdr`  ends up compiled as a call to memcpy:

```
int can_mcan_send(...)
{
...
	msg_ram->tx_buffer[put_idx].hdr = tx_hdr;
...
}
```
which copies improperly the tx_hdr into the msg_ram (verified with GDB).

I am using zephyr-sdk 0.13.1.

**To Reproduce**
Build a board which includes a compatible bosch,m-can-base, with either CONFIG_MINIMAL_LIBC_OPTIMIZE_STRING_FOR_SIZE=y
or
CONFIG_NEWLIB_LIBC=y

I am testing on board nucleo_g0b1re with the following device tree overlay:
```
/ {
	soc {
		can {
			compatible = "bosch,m-can-base";
			#address-cells = <1>;
			#size-cells = <1>;
			std-filter-elements = <28>;
			ext-filter-elements = <8>;
			rx-fifo0-elements = <3>;
			rx-fifo1-elements = <3>;
			rx-buffer-elements = <0>;
			tx-buffer-elements = <3>;

			can1: can@40006400 {
				compatible = "st,stm32-fdcan";
				#address-cells = <1>;
				#size-cells = <0>;
				reg = <0x40006400 0x400>, <0x4000B400 0x350>;
				reg-names = "m_can", "message_ram";
				interrupts = <21 0>, <22 0>;
				interrupt-names = "LINE_0", "LINE_1";
				status = "disabled";
				label = "CAN_1";
			};
			can2: can@40006800 {
				compatible = "st,stm32-fdcan";
				#address-cells = <1>;
				#size-cells = <0>;
				reg = <0x40006800 0x400>, <0x4000B750 0x350>;
				reg-names = "m_can", "message_ram";
				interrupts = <23 0>, <24 0>;
				interrupt-names = "LINE_0", "LINE_1";
				status = "disabled";
				label = "CAN_2";
			};
		};
	};
};
&can1 {
	pinctrl-0 = <&fdcan1_rx_pb8 &fdcan1_tx_pb9>;
	bus-speed = <500000>;
	sjw = <1>;
	sample-point = <875>;
	bus-speed-data = <500000>;
	sjw-data = <1>;
	sample-point-data = <875>;
	status = "okay";
};

```
With this configuration, sending a can frame (using `can_send`) with standard 11-bit ID 0x123, the ID sent on the bus is actually 0x101. There are also no data bytes sent.

**Expected behavior**
The correct CAN frame is sent.

**Impact**
Incorrect CAN frames are sent on the CAN bus.
I need to either change libc (sometimes impractical) or to patch myself the mcan driver.

**Environment:**
 - OS: Linux (ubuntu 21.10)
 - zephyr SDK 0.13.1
 - Zephyr 2.7.0

alexanderwachter, 2021-12-13:
Good catch!
Can you send a PR with a memcpy32 for 32 bits within m_can.c ?
IIRC there is already such a memcpy inside m_can.c

MJayat, 2021-12-13:
ok, I'll open a PR.

There's no memcpy32 available. I am not sure what the best implementation for this would be. Just casting to uint32_t* and looping over it would break C aliasing rules. (unless going through `char*`)

I suppose we can extend the use of unions with data_32 fields.

Just reading the code, I suspect that `msg_ram->std_filt[filter_nr] = filter_element;` in `can_mcan_attach_std` has the same problem too, but my testing was cut short by the first problem.

pdietl, 2021-12-13:
@MJayat We can do something like here: https://git.musl-libc.org/cgit/musl/tree/src/string/memcpy.c
and we can use the type attribute `__may_alias`

pdietl, 2021-12-15:
I will make a PR @MJayat if you're busy

MJayat, 2021-12-15:
Please do. I've been busy indeed.

I have just worked around it for now by changing libc :)

pdietl, 2021-12-15:
@MJayat can you try this?

henrikbrixandersen, 2021-12-21:
This was fixed by #41326. Closing.

PR #46182 (2022-06-11): [Backport v2.7-branch] drivers: can: m_can: fix alignmed issues
Backport https://github.com/zephyrproject-rtos/zephyr/pull/41326 because https://github.com/zephyrproject-rtos/zephyr/issues/41074 affects v2.7-branch

More background:
* mcan driver used to work in zephyr 2.6
* [This commit](https://github.com/zephyrproject-rtos/zephyr/commit/5d55730cf69dac7c8f99dcb17b04307cd050948c) landed into 2.7 
* MINIMAL_LIBC_OPTIMIZE_STRING_FOR_SIZE is set to yes when SIZE_OPTIMIZATIONS are enabled
* `memcpy` and `memset` versions used are not fit for unaligned memory access.

Fixes #41074
closed 2021-12-21 by henrikbrixandersen