zephyrproject-rtos/zephyr #41077: modem: gsm_mux: could not send more than 128 bytes of data on dlci

kurddt, 2021-12-10:
**Describe the bug**
When sending more than 128 bytes of data through a muxed uart the header is not correctly formatted

**Expected behavior**
The header should be correct when sending more than 128 bytes

**Impact**
It's possible to work-around this bug by splitting the data to send in smaller chunck

**Additional context**
This was experienced with while communicating with a ublox sara-r4 module. Probably the bug was hidden until #39776 was merged

kurddt, 2021-12-10:
@jukkar Since you seems to be the original author of the gsm muxing implementation I was wondering if you were able to test this use case (sending of more than 127 bytes)?
Since the 3GPP specification is not really explicit on which one of the length indicator byte is the MSB it could be a bug on the module side

jukkar, 2021-12-13:
> I was wondering if you were able to test this use case (sending of more than 127 bytes)?

I do not remember testing this. There are no unit tests for the muxing driver so all the testing is "manual" (at least now), there are probably bugs in the code because of this. Thanks for the PR, it looks good.

PR #41078 (2021-12-14): drivers: console: gsm_mux: fix length indicator
The MSB and LSB were inverted for length > 127
Fixes #41077
commit ba7e70e0 (2021-12-14): drivers: console: gsm_mux: fix length indicator

The MSB and LSB were inverted for length > 127
Fixes #41077
commit 02275d80 (2022-02-24): drivers: console: gsm_mux: fix length indicator

The MSB and LSB were inverted for length > 127
Fixes #41077
closed 2021-12-14 by MaureenHelm