zephyrproject-rtos/zephyr #41093: Kconfig.defconfig:11: error: couldn't parse 'default $(dt_node_int_prop_int,/cpus/cpu@0,clock-frequency)'

idkq, 2021-12-12:
Installed nRF command line tools + nRF connect + VS Code Extensions, followed https://www.youtube.com/watch?v=2cv_jjqk5hg&t=381s.

OS: Mac OS Monterey

Issue when building:

```
soc/arm/atmel_sam/Kconfig.defconfig:11: error: couldn't parse 'default $(dt_node_int_prop_int,/cpus/cpu@0,clock-frequency)': macro expanded to blank string
```

Print:

[image]

trond-snekvik, 2021-12-13:
Hi, this is most likely a configuration issue with your extension environment, and not an issue in the kconfig files themselves, so let's start there. 

This is usually a sign that the python executable running in your extension environment is different from the python executable west runs in. You can change this by configuring `kconfig.python` in your VS Code settings to point to the same python instance as west runs in. You'll need to restart VS Code for this to take effect.

idkq, 2021-12-13:
Thanks @trond-snekvik indeed my python was wrong. This solved the issue.

My environment $PATH (terminal) was pointing to a Espressif python. `/Users/me/.espressif/python_env/idf4.4_py3.9_env/bin/python`

If help others, how I fixed in VS Code (macOs):

1. Go to Code -> Preferences -> Settings
2. On the search bar at the top, type `kconfig`
3. On the left, below `Extensions` click `Kconfig`
4. On the right, find `Kconfig: Python` type the path including the executable name i.e. python
5. Close, restart

Since when I installed the toolchain, it was pointing to the 'wrong' python, should I reinstall it all again (any risk packages were installed in the wrong folder)? Thanks again

closed 2021-12-13 by idkq

trond-snekvik, 2021-12-13:
Great!

> Since when I installed the toolchain, it was pointing to the 'wrong' python, should I reinstall it all again (any risk packages were installed in the wrong folder)? 

If you struggle with the build, it should be the first check you make, but if `west build` still works fo you, I wouldn't bother

DrewOJensen, 2022-01-09:
Hey @trond-snekvik, I'm only digging up this thread because I'm having the exact same issue. I can build with west and it shows as being successful but the error continues to pop up. I wasn't sure if it was due to the python being used is within the toolchain rather than any python version installed locally.

This shows the same error message that @idkq got.
[image]

And here is the west build using the toolchain provided python 3.9.6, where as my python is the macOS Big Sur default of 2.7.
[image]

And just for proof, here is the build success verification printout.
[image]

Should I be pointing Kconfig: Python to the toolchain version to get rid of this error? I'm hesitant to do so because I don't want to "hard code" anything and screw up the environment defaults. It seems to be working just fine, I just wasn't sure if I can completely ignore that error or not. If so, it'd still be nice to clean the error up if possible.

trond-snekvik, 2022-01-09:
Hey, this looks like a python mismatch, yeah. Changing your "Kconfig: Python" should indeed fix this (you'll need to restart your vs code instance). It doesn't affect anything except the extension itself, and I'd be more wary of ignoring the error if you're using the extension's menuconfig UI to change things.

DrewOJensen, 2022-01-10:
Thanks for the confirmation @trond-snekvik. I'm not using the menuconfig UI, but none the less, I decided to put the toolchains Python link into the "Kconfig: Python" and the error has gone away. Thanks again for all your dedication!

smenor, 2022-04-17:
IDK if this is my issue or not but my Mac only has python3 and this doesn't work at all

trond-snekvik, 2022-04-18:
@smenor would you mind creating a post on Nordic's [devzone](https://devzone.nordicsemi.com) for this? Any details you can provide about your environment would be helpful.

smenor, 2022-04-19:
@trond-snekvik sure I can do that