zephyrproject-rtos/zephyr #41095: libc: newlib: 'gettimeofday' causes stack overflow on non-POSIX builds

binujacob1337, 2021-12-12:
**Describe the bug**
Calling 'gettimeofday' in a non-POSIX build results in recursive calls and ultimately causes stack overflow.

gettimeofday ->_gettimeofday_r -> _gettimeofday -> gettimeofday

Merging #35417 has introduced this issue. This merge has broken the fix added by #22508 for a similar recursion issue.

The fix in #35417 probably fixed the gettimeofday for POSIX builds, but has broken the non-POSIX builds.

**To Reproduce**
Steps to reproduce the behavior:

1. Modify prj.conf of samples/posix/gettimeofday to:

    CONFIG_MAIN_STACK_SIZE=2048
    CONFIG_NEWLIB_LIBC=y

2. Comment out sleep(1) in samples/posix/gettimeofday/src/main.c
3. west build -p always -b qemu_cortex_m3 samples/posix/gettimeofday
4. Check the build/zephyr/zephyr.lst to see the recursive calls as mentioned above.

**Logs and console output**

From zephyr.lst:

    00002c18 <gettimeofday>:
        2c18: 4b02       ldr r3, [pc, #8] ; (2c24 <gettimeofday+0xc>)
        2c1a: 460a       mov r2, r1
        2c1c: 4601       mov r1, r0
        2c1e: 6818       ldr r0, [r3, #0]
        2c20: f000 bcce b.w 35c0 <_gettimeofday_r>

    000035c0 <_gettimeofday_r>:
        35c0: b538       push {r3, r4, r5, lr}
        35c2: 2300       movs r3, #0
        35c4: 4d06       ldr r5, [pc, #24] ; (35e0 <_gettimeofday_r+0x20>)
        35c6: 4604       mov r4, r0
        35c8: 4608       mov r0, r1
        35ca: 4611       mov r1, r2
        35cc: 602b       str r3, [r5, #0]
        35ce: f7fd fddb bl 1188 <_gettimeofday>

    00001188 <_gettimeofday>:
        1188: b538       push {r3, r4, r5, lr}
        118a: 4604       mov r4, r0
        118c: 460d       mov r5, r1
        118e: 4804       ldr r0, [pc, #16] ; (11a0 <_gettimeofday+0x18>)
        1194: 4629       mov r1, r5
        1196: 4620       mov r0, r4
        1198: e8bd 4038 ldmia.w sp!, {r3, r4, r5, lr}
        119c: f001 bd3c b.w 2c18 <gettimeofday>

**Environment:**
 - OS: Ubuntu Linux
 - Toolchain: Zephyr SDK 0.13.1

**Fix suggestion**
Modifying the _gettimeofday in lib/libc/newlib/libc-hooks.c to call gettimeofday() only for POSIX builds may fix the problem.

    int _gettimeofday(struct timeval *__tp, void *__tzp)
    {
    #ifdef CONFIG_POSIX_API
	    return gettimeofday(__tp, __tzp);
    #else
	    return -1;
    #endif
    }

stephanosio, 2021-12-16:
> Modifying the _gettimeofday in lib/libc/newlib/libc-hooks.c to call gettimeofday() only for POSIX builds may fix the problem.

@binujacob1337 Since you have already suggested a fix for this, would you be interested in opening a PR for it?

binujacob1337, 2021-12-19:
@aurel32 Could you please provide the details of build environment that needed the fix in  #35417?

aurel32, 2021-12-19:
I used the Zephyr toolchain (version available around that time), on a SAM E70 board. That was using the posix/gettimeofday without any further changes. I am not sure I remember more details, sorry.

binujacob1337, 2021-12-19:
Thanks for the board info @aurel32!

PR #41276 (2022-01-21): libc: newlibc: Fix recursive gettimeofday() calls on non-Posix systems
Calling gettimeofday() from _gettimeofday() in a non-Posix build
environment can result in a recursive call loop, causing a stack
overflow. Modify _gettimeofday() to return -1 for non-posix systems
(the previous behaviour that was added in #22508).

Fixes #41095
commit 5fa5658e (2022-01-21): libc: newlibc: Fix recursive gettimeofday() calls on non-Posix systems

Calling gettimeofday() from _gettimeofday() in a non-Posix build
environment can result in a recursive call loop, causing a stack
overflow. Modify _gettimeofday() to return -1 for non-posix systems
(the previous behaviour that was added in #22508).

Fixes #41095