zephyrproject-rtos/zephyr #41097: west init issue

AshwinKrishn, 2021-12-12:
**Describe the bug**
When we do west init for the project, it tried to download the master branch. The branch is deleted. Why is the west tool not fetching main branch. 

Please also mention any information which could help others to understand
the problem you're facing:

**To Reproduce**
west init ~/zephyrproject

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Unable to do fresh west init 

**Logs and console output**

west init -mr ~/zephyrproject 
=== Initializing in /home/user/zephyrproject
--- Cloning manifest repository from r, rev. master
Initialized empty Git repository in /home/user/zephyrproject/.west/manifest-tmp/.git/
fatal: 'r' does not appear to be a git repository
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
ERROR: command exited with status 128: git fetch origin --tags -- master 'refs/heads/*:refs/remotes/origin/*'

**Environment (please complete the following information):**

No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.13 (stretch)
Release:	9.13
Codename:	stretch

**Additional context**

thedjnK, 2021-12-12:
1.  `west init ~/zephyrproject` and `west init -mr ~/zephyrproject` are not the same
2. What version of west are you using? This change was made to west half a year ago. https://github.com/zephyrproject-rtos/west/commit/f97f986f30ed54b3e5366cbac2e38c5ed7e38854

AshwinKrishn, 2021-12-12:
west version is:
West version: v0.6.3
The getting started page still has the "west init ~/zephyrproject". 

When I do "west init -mr ~/zephyrproject" the folowing happens

west init -mr ~/zephyrproject
=== Initializing in /home/user/zephyrproject
--- Cloning manifest repository from r, rev. master
Initialized empty Git repository in /home/user/zephyrproject/.west/manifest-tmp/.git/
fatal: 'r' does not appear to be a git repository
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
ERROR: command exited with status 128: git fetch origin --tags -- master 'refs/heads/*:refs/remotes/origin/*'

thedjnK, 2021-12-12:
0.6.3 is ancient 0.12.0 is latest. Update using `pip install -U west`

carlescufi, 2021-12-13:
As @thedjnK mentions, please update west (and Git if you have an old version of that too).

closed 2021-12-13 by carlescufi

AshwinKrishn, 2021-12-13:
Hello I did  upgrade west and now it works. But I had to manually install west package using wheel. Probably this is a very old machine,