zephyrproject-rtos/zephyr #41100: Non-standard RISC-V assembly is used

henry-hsieh, 2021-12-13:
In zephyr, some non-standard `jalr rd, rs` pseudo-instructions are used. The pseudo-instructions are not defined in official [RISC-V assembly manual](https://github.com/riscv-non-isa/riscv-asm-manual/blob/master/riscv-asm.md). I think that it's better not to use the instructions in kernel code and replace them to legal instructions such as `ret` for standard return pseudo-instruction or `jalr rd, rs, 0` for no offset jump register and link.

The non-standard instructions are listed below:
https://github.com/zephyrproject-rtos/zephyr/blob/78540881166a9634667fc646c3f5a685d8960db9/soc/riscv/riscv-privilege/common/soc_irq.S#L34
https://github.com/zephyrproject-rtos/zephyr/blob/78540881166a9634667fc646c3f5a685d8960db9/soc/riscv/riscv-privilege/common/soc_irq.S#L61
https://github.com/zephyrproject-rtos/zephyr/blob/d55c9b6d5470263abf1a3976336a29d34549a2a4/soc/riscv/esp32c3/soc_irq.S#L23
https://github.com/zephyrproject-rtos/zephyr/blob/02fceed732ab5fd8d2bdd9e2ebc80ba88a9c7184/soc/riscv/riscv-privilege/neorv32/soc_irq.S#L28
https://github.com/zephyrproject-rtos/zephyr/blob/cb307e2c64bc28cc91369ae5e7ca8d5c4236edb5/soc/riscv/riscv-ite/common/soc_irq.S#L60
https://github.com/zephyrproject-rtos/zephyr/blob/21d1ad3762302b3e461953df59430c77e0709274/arch/riscv/core/swap.S#L57
https://github.com/zephyrproject-rtos/zephyr/blob/4e0f7ea540a8b8c736c53ef6c0dd2b033ed3d021/arch/riscv/core/isr.S#L727

carlescufi, 2021-12-13:
@henry-hsieh thank you for pointing this out! Would you be willing to send a Pull Request with those fixes?

henry-hsieh, 2021-12-14:
Sure. I'll fix them later.

PR #41147 (2022-02-04): Fix non-standard pseudo-instructions of RISC-V
Fixes #41100.
commit 58d50a0e (2022-02-04): riscv: fix non-standard assembly of RISC-V

Non-standard `jalr rd, rs` pseudo-instructions are used.
This commit changes them to `ret` for standard return pseudo-instruction
or `jalr rd, rs, 0` for no offset jump register and link.

Fixes #41100.
commit 1e2cb821 (2023-03-17): riscv: fix non-standard assembly of RISC-V

Non-standard `jalr rd, rs` pseudo-instructions are used.
This commit changes them to `ret` for standard return pseudo-instruction
or `jalr rd, rs, 0` for no offset jump register and link.

Fixes #41100.

(cherry picked from commit 58d50a0e979b49d431e5a144b30787681d7a5cc7)
commit e87218a0 (2023-04-27): riscv: fix non-standard assembly of RISC-V

Non-standard `jalr rd, rs` pseudo-instructions are used.
This commit changes them to `ret` for standard return pseudo-instruction
or `jalr rd, rs, 0` for no offset jump register and link.

Fixes #41100.

(cherry picked from commit 58d50a0e979b49d431e5a144b30787681d7a5cc7)
closed 2022-02-04 by carlescufi