zephyrproject-rtos/zephyr #41117: Incorrect setting of gyro sensitivity in LSM6DSO driver

richardbarlow, 2021-12-13:
**Describe the bug**

The LSM6DSO driver does not set the gyro sensitivity (full-scale setting) correctly. An example walk-through of the issue when trying to set ±2000°/s full-scale:

Zephyr driver gyro configuration (hard-coded magic numbers that don't align with stmemsc API):

https://github.com/zephyrproject-rtos/zephyr/blob/13bf2e5e48737f5971bdfa2ee683453e379403ea/drivers/sensor/lsm6dso/lsm6dso.h#L67-L86

`LSM6DSO_DEFAULT_GYRO_FULLSCALE` used:

https://github.com/zephyrproject-rtos/zephyr/blob/9407add27c16f5a10eb1b508f8cb6a482bfc62ed/drivers/sensor/lsm6dso/lsm6dso.c#L782

Magic numbers passed through directly into stmemsc API:

https://github.com/zephyrproject-rtos/zephyr/blob/9407add27c16f5a10eb1b508f8cb6a482bfc62ed/drivers/sensor/lsm6dso/lsm6dso.c#L136-L146

Expected values for `lsm6dso_gy_full_scale_set` function call:

https://github.com/zephyrproject-rtos/hal_st/blob/575de9d461aa6f430cf62c58a053675377e700f3/sensor/stmemsc/lsm6dso_STdC/driver/lsm6dso_reg.h#L2775-L2784

The stmemsc API is such that the value of the `lsm6dso_fs_g_t` enum is directly used to set the `FS[1:0]_G` and `FS_125` fields in the `CTRL2_G` register. The values that the Zephyr driver are passing it appear to bear no relation to the `lsm6dso_fs_g_t` values and the end result is that, when requesting ±2000°/s full scale the gyro is actually configured for ±125°/s full scale:

 1. `LSM6DSO_DEFAULT_GYRO_FULLSCALE` == 3
 2. `lsm6dso_gy_full_scale_set` sets the `fs_g` member of the `lsm6dso_ctrl2_g_t` struct, which effectively results in a left shift by 1, setting the lower nibble to 6 (1010 in binary). Since the `FS_125` bit is set, the value of the `FS[1:0]_G` field is ignored by the IMU.

avisconti, 2021-12-15:
@richardbarlow If you want to take a look to the fix I would appreciate it. Thanks very much for reporting this bad issue.

richardbarlow, 2021-12-15:
@avisconti Thank you for fixing the problem so promptly! I am not in a position to test the fix on real hardware at the moment, but reading through the change it looks good :+1:

I'll try to get the fix tested in the next couple of days.

avisconti, 2021-12-15:
> @avisconti Thank you for fixing the problem so promptly! I am not in a position to test the fix on real hardware at the moment, but reading through the change it looks good +1
> 
> I'll try to get the fix tested in the next couple of days.

Thanks very much!

PR #41241 (2021-12-18): LSM6DSO accel/gyro sensor: Add DTS property for range/odr/power-mode
LSM6DSO sensor (both accel and gyro)                                                                              
                                        
Move odr/range properties from Kconfig to DTS. Moreover add power-mode properties as well.                                                                                
Fixes #41117
commit 18e8cada (2021-12-18): drivers/sensor: lsm6dso: Move odr/range Kconfig property into dts

Move odr and range properties for both accelerometer and gyro
from Kconfigs to Device Tree.
Fixes #41117
closed 2021-12-18 by cfriedt