zephyrproject-rtos/zephyr #41153: rt i2s build fail

lgl88911, 2021-12-14:
**Describe the bug**
rt1052/1062 i2s compilation error

**To Reproduce**
Steps to reproduce the behavior:
1. download test branch https://github.com/lgl88911/zephyr/tree/rt_i2s_verify
2. git checkout 17b447087f50
3. west build -b mimxrt1060_evk zephyr/samples/drivers/i2s/echo/
4. See error
```
/mnt/e/westz/zephyrmaster/zephyr/drivers/i2s/i2s_mcux_sai.c:72:11: error: duplicate member 'clk_src'
   72 |  uint32_t clk_src;
      |           ^~~~~~~
/mnt/e/westz/zephyrmaster/zephyr/drivers/i2s/i2s_mcux_sai.c: In function 'audio_clock_settings':
/mnt/e/westz/zephyrmaster/zephyr/drivers/i2s/i2s_mcux_sai.c:991:10: error: switch quantity not an integer
  991 |  switch (dev_cfg->clk_sub_sys) {
      |          ^~~~~~~
In file included from /mnt/e/westz/zephyrmaster/zephyr/include/sys/util_macro.h:34,
                 from /mnt/e/westz/zephyrmaster/zephyr/include/sys/atomic.h:16,
                 from /mnt/e/westz/zephyrmaster/zephyr/include/kernel_includes.h:21,
                 from /mnt/e/westz/zephyrmaster/zephyr/include/kernel.h:17,
                 from /mnt/e/westz/zephyrmaster/zephyr/drivers/i2s/i2s_mcux_sai.c:15:
/mnt/e/westz/zephyrmaster/zephyr/drivers/i2s/i2s_mcux_sai.c: At top level:
/mnt/e/westz/zephyrmaster/zephyr/drivers/i2s/i2s_mcux_sai.c:1103:47: error: expected '}' before ';' token
 1103 |   .clk_pre_div = DT_INST_PROP(i2s_id, pre_div);  \
      |                                               ^
```

There are several obvious clerical errors in i2s_mcux_sai.c that cause the compilation to fail. Maybe https://github.com/zephyrproject-rtos/zephyr/pull/37335 has not been submitted to the final version of i2s_mcux_sai.c.
Although I fixed the compilation error in https://github.com/lgl88911/zephyr/commit/f2cc22b6df794a697ba35df478dcb046ed96a63d, it seems that there are still some pinmux-related issues that need to be resolved. Can you commit the code that you passed the test again? Thanks!

hakehuang, 2021-12-14:
@lgl88911 this driver still need some more PR to work, but there is some merge/split error, I will provide a quick fix

lgl88911, 2021-12-25:
@hakehuang I cherry-pick 2cbdd12e9cb1fba279dc82a1227aed50cf3ace1d, build pass. Thank you for your quick fixed.

closed 2021-12-25 by lgl88911