zephyrproject-rtos/zephyr #41218: LE Audio: TBS/CCP long read

Thalley, 2021-12-15:
Need to properly handle long notifications from TBS in the Call Control client. 

The easy way is to simply do a read if the value matches the MTU, but is slightly troublesome as the CC client does not now whether notify multiple, or notify (1 octet difference in header len) was used, and can thus not properly determine if the lenght is the MTU size. 

Request to add a field to the long values for the clients to easily know if they have been truncated has been sent to the GA group, and awaiting response. The implemenation for this may depend on the response.

kruithofa, 2024-01-30:
Related to #68173

Thalley, 2024-03-06:
Closing in favor of https://github.com/zephyrproject-rtos/zephyr/issues/68173

closed 2024-03-06 by Thalley