zephyrproject-rtos/zephyr #41220: STM32H7: Check for VOSRDY instead of ACTVOSRDY

benediktibk, 2021-12-15:
**Describe the bug**
In soc_m7.c:91the system is waiting for VOSRDY to become ready. According to the reference manual of a STM32H735 this bit should already be ready anyway at this point, but instead one should wait for ACTVOSRDY to become ready.

For working on a different issue I already changed this locally, and honestly it made no difference. At least in my case. I just stumbled upon this by accident and I think the currently implemented behaviour is incorrect.

**To Reproduce**
Steps to reproduce the behavior:
1. Run an example on STM32H735 Discovery Kit

**Expected behavior**
Waiting for ACTVOSRDY to be ready instead of VOSRDY.

**Impact**
None so far.

**Logs and console output**

**Environment (please complete the following information):**
Zephyr Version 38c42f69d28b8ce7cefb2b1a265400e83dd654aa

**Additional context**

benediktibk, 2021-12-15:
Just checked the reference manuals for the (as far as I know) all by zephyr supported H7 variants:

- H723
- H735
- H743
- H745
- H747
- H750
- H753

This behaviour should be the same for all of them, hence no differentiation based on the chip will be necessary.

FRASTM, 2021-12-16:
In the stm32h7_init(), as the LDO  (is switched on for M7 run Mode, it seems correct to monitor the VOSRDY bit of the PWR_D3CR register for the Domain3 : that's the LL_PWR_IsActiveFlag_VOS()
My understanding is that The ACTVOSRDY bit of the PWR_CSR1 register should be used to monitor the SVOS level as set by the PWR_CR1 register --> VOS in STOP mode.

benediktibk, 2021-12-16:
[image]
[image]

Looking at the reference manual I come to a different conclusion. VOSRDY will happen way earlier, before the actual code is executed after step 4 of the startup sequence. The application running on the chip should then check for ACTVOSRDY before the VOS levels can be changed.

benediktibk, 2021-12-21:
Just stumbled upon this: https://github.com/zephyrproject-rtos/hal_stm32/blob/34fe6f7332925417b5c7fbd1eed2414b2741c2e4/stm32cube/stm32h7xx/drivers/src/stm32h7xx_hal_pwr_ex.c#L350

Looks like HAL is also checking ACTVOSRDY.

benediktibk, 2021-12-23:
I addressed the issue in #41439

FRASTM, 2022-02-14:
now addressed by https://github.com/zephyrproject-rtos/zephyr/pull/41871

closed 2022-04-19 by benediktibk