zephyrproject-rtos/zephyr #41222: tests: remove not existing platforms from platform allow or integration_platform

gopiotr, 2021-12-15:
**Describe the bug**
I worked on checking of platform existence in this PR: #41081 and in the result I discovered, that among Zephyr tests and samples, there are some not existing platforms in `platform_allow` and `integration_platforms` lists.

I tried to collect all those cases and assign to them person from CODEOWNERS file or from commits history.

People who are mention in following list: please consider if following not existing platforms should be removed from `platform_allow` or `integration_platforms` list or should be replaced by other platform. If some of you feel, that you are not responsible for some of those test please let me know about it - I apologize in advance.

**`platform_allow`**
- [ ] `tests/subsys/storage/stream/stream_flash` | `storage.stream_flash.mpu_allow_flash_write` | `nrf52840_pca10056` | @hakonfam
- [ ] `tests/drivers/build_all/ieee802154` | `ieee802154.build.cc12xx_cc2520` | `cc1352_sensortag` | @JordanYates
- [ ] `tests/kernel/workq/critical` | `kernel.workqueue.nsim` | `nsim_em_mpu_stack_guard` | @nashif @dcpleung @andyross
- [ ] `tests/posix/common` | `portability.posix.common.nsim` `portability.posix.common.newlib.nsim` | `nsim_em_mpu_stack_guard` | @dcpleung @pfalcon
- [ ] `samples/subsys/ipc/openamp` | `sample.ipc.openamp` | `v2m_musca` | @galak
- [ ] `samples/userspace/syscall_perf` | `sample.syscall_performances` | `riscv32` | @dcpleung @nashif
- [ ] `samples/drivers/counter/maxim_ds3231` | `sample.basic.maxim_ds3231` | `nrf51_pca10028` | @pabigot
- [ ] `samples/boards/arc_secure_services` | `sample.board.arc_secure_services` | `em_starterkit_em7d_secure` | @ioannisg 
- [ ] `samples/tfm_integration/tfm_secure_partition` | `sample.tfm.secure_partition` `sample.tfm.secure_partition.ipc` | `nrf9160dk_nrf9160ns` `v2m_musca_s1_nonsecure` `nrf5340dk_nrf5340_cpuappns` `mps2_an521_nonsecure` | @microbuilder @oyvindronningstad

**`integration_platforms`**
- [ ] `tests/subsys/jwt` | `libraries.encoding.jwt` | `frdm_k64` | @ceolin
- [ ] `tests/drivers/flash` | `drivers.flash.stm32` | `nucleo_f103rc` | @str4t0m 
- [ ] `samples/subsys/ipc/rpmsg_service` | `sample.ipc.rpmsg_service.nrf` | `nrf5340dk_nrf5340_cpuapp bl5340_dvk_cpuapp` | @lairdjm - please convert this into  list of dashes
- [ ] `samples/subsys/ipc/ipc_service/static_vrings_mi` | `sample.ipc.static_vrings_mi.nrf` | `nrf5340dk_nrf5340_cpuapp bl5340_dvk_cpuapp` | @carlocaione - please convert this into  list of dashes

**Expected behavior**
All not existing platforms should be removed or replaced by other platform.

**Environment**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA used: 92fbd7da14d1c0c69457441fb13838d0b81f7f12

pfalcon, 2021-12-15:
> tests/posix/common | portability.posix.common.nsim portability.posix.common.newlib.nsim | nsim_em_mpu_stack_guard | @dcpleung @pfalcon

I'd say that in this case (and perhaps others) you should ping maintainers of a platform in question, not the code piece where it's mentioned (well, maybe @dcpleung is a maintainer of "nsim" stuff, but I know nothing about it).

Doing a quick git log grep, adhoc nsim test configs above were introduced in 7387b56c61cd28285f8174e201f94309de446f8f . AFAIK, Andrew Boie no longer works on the project, so probably makes no sense to ping him. But from the above, seems like NSIM is an Intel thing, so you may want to ping Intel folks about it (e.g. @nashif if nobody else).

oyvindronningstad, 2021-12-15:
For TFM you can ping @SebastianBoe instead of me.

gopiotr, 2021-12-15:
> I'd say that in this case (and perhaps others) you should ping maintainers of a platform in question, not the code piece where it's mentioned (well, maybe @dcpleung is a maintainer of "nsim" stuff, but I know nothing about it).
> 
> Doing a quick git log grep, adhoc nsim test configs above were introduced in 7387b56 . AFAIK, Andrew Boie no longer works on the project, so probably makes no sense to ping him. But from the above, seems like NSIM is an Intel thing, so you may want to ping Intel folks about it (e.g. @nashif if nobody else).

@pfalcon Thank you for your comment. Yes you are right, in this case I will contact with @nashif, who merged Andrew Boie commit with `nsim_em_mpu_stack_guard` platform. Sorry to inconvenience you.

SebastianBoe, 2021-12-15:
@gopiotr : For the tfm related boards.

The board names should be changed from nonsecure and ns to _ns.

nashif, 2021-12-15:
this is one to do that, ping everyone in the project about fixing what counts as a typo or misnamed platforms, OR you can take a stab at it and do the changes and fixes to get the right platforms in, the latter, will be faster and more efficient, you can then add people for review to make sure you have the right name.

gopiotr, 2021-12-15:
@nashif thank you for this PR: #41238
BTW do you know what platform should be used there: `samples/userspace/syscall_perf` - `sample.syscall_performances` instead of `riscv32`? I tried to use `qemu_riscv32` board, but there is an error during build this sample on this board.

nashif, 2021-12-15:
> @nashif thank you for this PR: #41238
> BTW do you know what platform should be used there: `samples/userspace/syscall_perf` - `sample.syscall_performances` instead of `riscv32`? I tried to use `qemu_riscv32` board, but there is an error during build this sample on this board.

no, this sample seems to be broken, we need to file a separate bug. @amergnat can you please fix this?

PR #41238 (2021-12-17): tests: fix typos and misnamed platforms
Various obsolote and misnamed platfomrs in test filters theat went
undetected for a while.

Fixes #41222
commit 05ecd46a (2021-12-17): tests: fix typos and misnamed platforms

Various obsolote and misnamed platfomrs in test filters theat went
undetected for a while.

Fixes #41222
closed 2021-12-17 by nashif