zephyrproject-rtos/zephyr #41228: LE Audio: Add a codec to Zephyr

Thalley, 2021-12-15:
zephyr-bt host meeting 2020-09-08
From discussion at IOP in Vienna.
It is possible to get access to LC3 source for EHIMA - shared on individual basis.
MIT license, going into AOSP at some point
Non-optimized codec

asbjornsabo commented on 23 Mar:
EHIMA LC3 codec contributed
https://github.com/zephyrproject-rtos/liblc3codec/

Thalley, 2021-12-15:
@Casper-Bonde-Bose @rymanluk Was this something you had started looking into?

Casper-Bonde-Bose, 2021-12-15:
@Thalley Yes, I have it building and have made c++ wrappers for Lc3Config and Lc3Encoder classes and have integrated the use of the codec into the unicast_audio_client sample application to transmit a 500Hz tone rather than test data (for the native_posix build). I'm experiencing some issues with encryption which means I cannot verify the encoding using an air-trace. Also I have a timing issue - even though I have a timer firing every 10ms it seems to fire every 20ms - hence the data rate is only half of the expected.
I'll try to disable encryption to validate the content of the CIS stream a nd have an idea of how to fix the timing issue.
I'll upload the Zephyr module build additions to https://github.com/zephyrproject-rtos/liblc3codec/ when I have validated it works. When that is approved and merged I can add the module to Zephyr and upload the integration into the sample-applications
Initially I'll only add support for the native_posix build and let others extend to other relevant build-targets - the LC3 codec is not optimized, hence I'm not sure it will perform well on an embedded platform.

Thalley, 2021-12-15:
@Casper-Bonde-Bose Can we assign this to you then, and move to in-progress?

Casper-Bonde-Bose, 2021-12-15:
@Thalley Sure. I assigned it to me, but I don't think I have access to move to in progress :-) At least I cannot drag the issue in the project overview.

Thalley, 2021-12-15:
> @Thalley Sure. I assigned it to me, but I don't think I have access to move to in progress :-) At least I cannot drag the issue in the project overview.

Interesting. I'll have to look further into that. Moved it into in-progress for you :)

Casper-Bonde-Bose, 2022-01-11:
Zephyr module PR (needs merge before Zephyr PR can build)
https://github.com/zephyrproject-rtos/liblc3codec/pull/4
Doing cleanup of the Zephyr PR, will upload draft version shortly allowing transfer of an LC3 encoded sinus tone over a CIS link.

Casper-Bonde-Bose, 2022-02-08:
The changes are pending review from a second reviewer:
First to be submitted:
https://github.com/zephyrproject-rtos/liblc3codec/pull/4

Then this:
https://github.com/zephyrproject-rtos/zephyr/pull/42443
But the latter requires an update in the manifest to point to the commit-id of the LC3 PR before it is merged.

PR #44225 (2022-03-30): modules: Add zephyr lc3 codec
Sample code demonstrating how the LC3 codec can be used for encoding and decoding audio data.
This is a simple integration into the existing BAP unicast application that uses a generated sine wave as audio data, encodes it and transfers it over a CIS. The receiver decodes the data and discards it after checking for errors.
Fixes #41228 
This replaces https://github.com/zephyrproject-rtos/zephyr/pull/42443 using the new c-version of the LC3 codec
closed 2022-03-30 by carlescufi