zephyrproject-rtos/zephyr #41237: drivers: ieee802154_dw1000: use dedicated workqueue

cfriedt, 2021-12-15:
**Describe the bug**

Driver has dedicated workqueue for IRQ processing.
Submit work to dedicated workqueue intead of system workqueue.
It also fixes driver functionality when NET_TC_TX_COUNT is set to 0.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

cfriedt, 2021-12-15:
@jfischer-no - can you please fill in the details above?

jfischer-no, 2022-02-04:
Added more details to bug description.

closed 2022-02-04 by carlescufi