zephyrproject-rtos/zephyr #41240: logging can get messed up when messages are dropped

wcappelle, 2021-12-15:
**Describe the bug**
When using USB logging, it might happen that the logging is not behaving like it should. The behavior is observed that when there are dropped messages, the logging only outputs messages (flushing) on the uart, when there are <dropped messages> + 1 in the ringbuffer.
The dropped messages in my case are occurring at startup (I only see the dropped message info the moment I start minicom on the usb ACM device). 

I have this debugged further to the buffered_cnt in log_core.c which can have a negative value when there are dropped frames over UART. Once this happens, you need to log the number of dropped frames, before the z_log_msg_post_finalize function acts again to what is is designed for. So it's miscounting the number of messages actually in the buffer.

sninppet of my conf:
```
# Shell
CONFIG_SHELL=y
CONFIG_SHELL_BACKEND_SERIAL=y
CONFIG_SHELL_BACKEND_SERIAL_RX_RING_BUFFER_SIZE=256
CONFIG_SHELL_BACKEND_SERIAL_TX_RING_BUFFER_SIZE=256
# Logging
CONFIG_LOG=y
CONFIG_LOG_FUNC_NAME_PREFIX_ERR=y
CONFIG_LOG_FUNC_NAME_PREFIX_WRN=y
CONFIG_LOG_FUNC_NAME_PREFIX_INF=y
CONFIG_LOG_STRDUP_MAX_STRING=16
CONFIG_LOG_STRDUP_BUF_COUNT=64
CONFIG_LOG_BUFFER_SIZE=2048
CONFIG_PRINTK=y
# USB driver & stack
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_DRIVER=y
# Logging over usb backend
CONFIG_SERIAL=y
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=y
CONFIG_UART_LINE_CTRL=y
CONFIG_UART_ASYNC_API=y
```

usb overlay:
```
/ {
    chosen {
        zephyr,console = &cdc_acm_uart0;
        zephyr,shell-uart = &cdc_acm_uart0;
    };
};

&zephyr_udc0 {
    status = "okay";
    cdc_acm_uart0: cdc_acm_uart0 {
        compatible = "zephyr,cdc-acm-uart";
        label = "CDC_ACM_0";
    };
};
```

this has been tested on a stm32l562e_dk board with a custom app. 

**To Reproduce**

**Expected behavior**

**Impact**

logging is deferred up to the number of messages which have been dropped (always)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain zephyr sdk-13.2
 - zephyr version: 7b86137b6cca28c3d9c5f96b84a4408de198c8de (main of yesterday afternoon 14 dec )

**Additional context**

Following code is avoiding the problem, but i'm sure it's not the correct solution since the check is not atomic, and is more polishing than fixing the incorrect decrement (therefore not a PR, but a bug report).
```
diff --git a/subsys/logging/log_core.c b/subsys/logging/log_core.c
index c60ac48715..fb91494479 100644
--- a/subsys/logging/log_core.c
+++ b/subsys/logging/log_core.c
@@ -808,7 +808,10 @@ bool z_impl_log_process(bool bypass)
 
        msg = get_msg();
        if (msg.msg) {
-               atomic_dec(&buffered_cnt);
+               if(buffered_cnt>0) {
+                       atomic_dec(&buffered_cnt);
+               }
+
                msg_process(msg, bypass);
        }
 
```

wcappelle, 2021-12-23:
Some more info: this problem is not limited to USB logging but also affects normal logging (UART). It can occur the moment you have dropped messages (not linked to usb , or startup of it).

nordic-krch, 2021-12-29:
@wcappelle can you validate provided fix?

wcappelle, 2021-12-29:
@nordic-krch , thanks for looking into fixing this issue. i have currently no access to any devices. I will validate this 2nd week of january when back from holidays

nordic-krch, 2021-12-29:
@wcappelle sure, fyi i managed to reproduce the issue.

PR #41490 (2022-01-05): logging: Fix counting of buffered messages
When message is dropped then log_process is called with
bypass flag set and additionally z_log_dropped() is called.
In both functions counter of buffered messages was decremented.
That resulted in counter being decremented twice. It resulted
in logging misbehavior after messages being dropped (delayed
processing). Fixing it by decrementing the counter in log_process
only when bypass flag is not set.

Additionally, waking up algorithm is improved by using value returned by `atomic_inc`
instead of reading variable directly.

Fixes #41240.
Fixes #41488.
closed 2022-01-05 by carlescufi