zephyrproject-rtos/zephyr #41249: add ST: BLUENRG-LP HAL module

manoj153, 2021-12-15:
## Introduction

There is some community interest to get ST bluenrg-lp SoC supported. #39520

This RFC is to understand the required work in a systematic order.

### Problem description
#### Based on my assessment and a discussion in discord :

- Required to add vendor HAL, [STSW-BNRGLP-DK](https://www.st.com/content/my_st_com/en/products/embedded-software/evaluation-tool-software/stsw-bnrglp-dk.html#get-software)

- inconvenient is the folder compromise of vendor codes(HAL), and samples  will be available  only after going through GUI installation steps (.exe)

#### Following is the extracted tree (that I found relevant)

```
|-- Drivers
|   |-- BSP
|   |   |-- Common
|   |   |-- Components
|   |   `-- STEVAL-IDB011V1
|   |-- CMSIS
|   |   |-- Device
|   |   `-- Include
|   |-- External_micro
|   |   |-- BSP
|   |   |-- CMSIS
|   |   `-- STM32L4xx_HAL_Driver
|   `-- Peripherals_Drivers
|       |-- inc
|       `-- src
```
Can I keep the directory structure intact as the vendor leaves, or can someone experience suggest?

erwango, 2021-12-16:
I'd suggest to keep only the minimum required for zephyr porting: the peripheral drivers and CMSIS.
(I assume that the `STM32L4xx_HAL_Driver` mentioned relates to an additional STM32L4 mcu available on the board).
Also, I suggest to push these to https://github.com/zephyrproject-rtos/hal_st.

manoj153, 2021-12-16:
Follow the directory structure as in `hal_stm32`?

nashif, 2023-04-13:
stale, please reopen if needed with updated information.