zephyrproject-rtos/zephyr #41255: drivers/can/can_mcan.c: address-of-packed-member warnings

erwango, 2021-12-16:
**Describe the bug**
Since cf111962e0d0c388bf806bbd9f17d49732daa264 warnings are generated on "address-of-packed-member"

3 of them are generated when compiling `drivers/can/can_mcan.c`
```
/local/mcu/zephyrproject/zephyr/drivers/can/can_mcan.c: In function 'can_mcan_init':
/local/mcu/zephyrproject/zephyr/drivers/can/can_mcan.c:389:2: error: converting a packed 'struct can_mcan_msg_sram' pointer (alignment 1) to a 'uint32_t' {aka 'unsigned int'} pointer (alignment 4) may result in an unaligned pointer value [-Werror=address-of-packed-member]
  389 |  for (uint32_t *ptr = (uint32_t *)msg_ram;
      |  ^~~
In file included from /local/mcu/zephyrproject/zephyr/drivers/can/can_mcan.c:11:
/local/mcu/zephyrproject/zephyr/drivers/can/can_mcan.h:148:8: note: defined here
  148 | struct can_mcan_msg_sram {
      |        ^~~~~~~~~~~~~~~~~
/local/mcu/zephyrproject/zephyr/drivers/can/can_mcan.c: In function 'can_mcan_get_message':
/local/mcu/zephyrproject/zephyr/drivers/can/can_mcan.c:525:15: error: taking address of packed member of 'struct can_mcan_rx_fifo' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  525 |    for (src = fifo[get_idx].data_32,
      |               ^~~~
/local/mcu/zephyrproject/zephyr/drivers/can/can_mcan.c: In function 'can_mcan_send':
/local/mcu/zephyrproject/zephyr/drivers/can/can_mcan.c:699:9: error: taking address of packed member of 'struct can_mcan_tx_buffer' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  699 |   dst = msg_ram->tx_buffer[put_idx].data_32,
      |         ^~~~~~~
```

**To Reproduce**
Could be reproduced with the following command:
```
twister -b -p nucleo_g474re -s tests/subsys/canbus/isotp/implementation/can.isotp.implemmentation -N
```

**Expected behavior**
No warnings at compile time

erwango, 2021-12-16:
@alexanderwachter would you mind having a look ?

alexanderwachter, 2021-12-16:
@pdietl is currently working on it

henrikbrixandersen, 2021-12-16:
@pdietl Please post a PR as soon as possible as this is causing CI failures.

henrikbrixandersen, 2021-12-16:
Raising priority since this is causing CI failures.

henrikbrixandersen, 2021-12-21:
This was fixed in #41326. Closing.

closed 2021-12-21 by henrikbrixandersen