zephyrproject-rtos/zephyr #41256: Zero Latency Interrupts conflicts

caco3, 2021-12-16:
We have an application based on a nrf52840 and zephyr 2.7.0.
The system is used for a measurement device. We run an external ADC at 10 kSamples/s and read its data through SPI DMA transfers. 
Thus, the ADC triggers an PPI interupt every 100 us which gets directly processed by the DMA without CPU interactions.
And also a timer counter gets incremented through PPI.
After 30 samples (=> 333 Hz) the timer counter trigger an Interrupt Handler which toggles the DMA buffer and copies the data into a FIFO. The SPI and the Timer are connected via PPI.

We discovered that the timer counter interrupt latency sometimes is higher than the SPI interrupt interval (100 us), causing an overflow of the DMA buffers. Therefore we enabled `CONFIG_ZERO_LATENCY_IRQS=y` and started to use `IRQ_DIRECT_CONNECT` with priority 0 and `IRQ_ZERO_LATENCY` enabled:
```C
void setup(void) {
    [..]
    IRQ_DIRECT_CONNECT(TIMER2_IRQn, 0, TIMER2_IRQHandler, IRQ_ZERO_LATENCY);
}

ISR_DIRECT_DECLARE(TIMER2_IRQHandler) {
    NRF_TIMER2->EVENTS_COMPARE[0] = 0;

    /* Toggle buffers */
    if(adc_dma_buffer_toggle == 0){ // First Buffer was selected => switch to second buffer
        NRF_SPIM3->RXD.PTR = (uint32_t) &adc_dma_buffer1; // Switch to the Second DMA Buffer
    }
    else { // Second Buffer was selected => switch to first buffer
        NRF_SPIM3->RXD.PTR = (uint32_t) &adc_dma_buffer0; // Switch to the First DMA Buffer

    // Copy data from inactive buffer to FIFO
    [..]

    return 1;
}
```

This seems to fix the interrupt latency issue if run standalone.
How ever we also need to transfer the FIFO data over BLE in the 2MB PHY Mode.
Once we started to use the above interrupt configuration with `IRQ_ZERO_LATENCY`, we started to get BLE Disconnects with reason `0x08` (`Connection Timeout`). We already increased the `Connection supervision timeout` to `500` (`5000ms`) but still get those timeouts.

Then we additionally enabled `CONFIG_BT_CTLR_ZLI=y`, but this still causes the disconnects and sometimes even Seg Faults after a while: `ASSERTION FAIL [!radio_is_ready()] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c:324`.

When **not** using interrupts with  `IRQ_ZERO_LATENCY`, BLE runs stable and we didn't see disconnects (but the DMA buffer overflows due to the too high timer counter interrupt latency).

Is it possible that BLE and low latency interrupts can not be used at the same time?
What would be a possible solution?

- Zephyr: `2.7.0`
- SDK: `0.13.1`
- We are not using any `irq_locks` (As warned in https://github.com/zephyrproject-rtos/zephyr/issues/40298).
-  We have other I2C communication at the same time, but without interrupts and thus with lower priorities.
- `CONFIG_BT_CTLR_ADVANCED_FEATURES=y` are active as well
- `CONFIG_BT_MAX_CONN` defaults to 1

cvinayak, 2021-12-16:
Under default interrupt configurations (no ZLI), Zephyr Bluetooth Controller uses zephyr interrupt priority level of 0 for Radio ISR, and share its bottom halves with all other SoC peripherals which use level of 1 (refer to dts files for default values).

If using ZLI, then Controller Radio ISR and any application set ISR use the same ZLI priority level, and hence Controller will not be able to tolerate ISR latencies of ~100 us introduced by other ISRs.

As ZLI cannot use all the Kernel features, please try to not use ZLI if your application can use Zephyr interrupt priority value > 1 for application used SoC peripherals.
ZLI be used only when application threads would introduce ISR latencies due to IRQ locking.

Zephyr Bluetooth Controller will use priorities 0 and 1, application used SoC peripherals can use 2, 3 etc...

caco3, 2021-12-17:
Dear @cvinayak 

Thank you for your explanations!
I am not sure if I yet understood it fully.
I have some further questions:
1. What means `bottom halves`?
2. You write `If using ZLI`. Do you mean the usage of `IRQ_ZERO_LATENCY` with this? Or the activation of one of the two flags `CONFIG_BT_CTLR_ZLI` or `CONFIG_ZERO_LATENCY_IRQS`? According to the docs the first one anyway depends on the 2nd one.

I tested it once more:
```
CONFIG_ZERO_LATENCY_IRQS=y
CONFIG_BT_CTLR_ZLI=y
```
Case 1 (No ZLI for the application):
> `IRQ_DIRECT_CONNECT(TIMER2_IRQn, TIMER_IRQ_PRIORITY, TIMER2_IRQHandler, 0);`
With `TIMER_IRQ_PRIORITY` = 0
Result: `100us` Interupt Latency not garanteed => lose of data

Case 2 (Also ZLI for the application):
> `IRQ_DIRECT_CONNECT(TIMER2_IRQn, TIMER_IRQ_PRIORITY, TIMER2_IRQHandler, IRQ_ZERO_LATENCY);`
With `TIMER_IRQ_PRIORITY` = 0, 1, 2, 3
Result: No data lose, but `ASSERTION FAIL [!radio_is_ready()] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c:324` after a while.

In the end, does that mean that, when using BLE (with the given bluetooth config), does not allow to have application interrupt latency < `100us`?

cvinayak, 2021-12-17:
> What means bottom halves?

Controller does post-processing of PDUs and manages radio event scheduling etc, in a lower priority execution context. This is the bottom half of Radio H/w ISR (which is the top-half).

> You write If using ZLI. Do you mean the usage of IRQ_ZERO_LATENCY with this? Or the activation of one of the two flags CONFIG_BT_CTLR_ZLI or CONFIG_ZERO_LATENCY_IRQS? According to the docs the first one anyway depends on the 2nd one.

By `if using ZLI`, I mean `CONFIG_ZERO_LATENCY_IRQS=y`, `CONFIG_BT_CTLR_ZLI=y` and application defined ISR(s) use `IRQ_ZERO_LATENCY` 

> In the end, does that mean that, when using BLE (with the given bluetooth config), does not allow to have application interrupt latency < 100us?

Zephyr Bluetooth Controller's bottom half ISRs consume > 100 us, but the Radio ISR (priority value 0)  consume < 100us. You could lower this `CONFIG_BT_CTLR_ULL_HIGH_PRIO` to 2 and keep your timer priority higher that the Controller's bottom half priorities at value 1.

caco3, 2021-12-17:
Dear @cvinayak

Thank you once more for your support!

I think I understood it now enough.
I adjusted the priorities according your suggestion and I do now no longer have the data lose.

I still get occationally a `ASSERTION FAIL [!radio_is_ready()] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c:324`. I think this is because my interrupt handler is to slow and I am spending too much time in it, causing the `lll_conn_isr_rx()` starves.

I tried to measure its time consumption with following code:
```
ISR_DIRECT_DECLARE(TIMER2_IRQHandler) {
    uint32_t t0 = k_cycle_get_32();

    [..]

    LOG_WRN("%d", (k_cycle_get_32() - t0) * 1000000 / sys_clock_hw_cycles_per_sec());
    return 1;
}
```

I got timmings of up to `152 us` which is really a lot.
I expect that this measurement is not very accurate and will repeat it with an oscilloscope.

cvinayak, 2021-12-18:
> my interrupt handler is to slow

Please keep your timer interrupt priority level lower, i.e. a priority value greater than 1, so that radio ISR priority value is kept at 0. This way Radio ISR can pre-empt your timer interrupt. I believe Radio ISR should not add more than 50 us latency for the timer ISR keep at priority value  greater than 1.

cvinayak, 2021-12-18:
> my interrupt handler is to slow

Are you sure you have set the zephyr priority value for your timer to 1  and not using `IRQ_ZERO_LATENCY` ?
Radio ISR should be at zephyr priority value of 0, and pre-empt your timer ISR, this way Radio ISR should not be impact by the timer ISR duration.

Try using these:

caco3, 2021-12-21:
Dear @cvinayak

You were right, I still had my timer flagged with `IRQ_ZERO_LATENCY`. I switched the flag now back to `0` and had no more BLE Assert crashes.
It runs now stable (as without crashing), how ever I still see occational data lose. Several times I saw now that my timer handler got delayed by `10.5ms` (!), althrough I need it to be called within `100us`. I will show the relevant snippets below, maybe you see something I still do wrong.
Also for easier understanding, I made a drawing:
[image]

We have an ADC which fires its DRDY (Dtata Ready) line every 100us.
To that signal we connected the SPIM3 peripheral in array mode.
To the same signal we connect the Timer2 counter
Once the timer counter reaches 30 (meaning we got 30 ADC interrupts and the SPPIM3 transfered 30 samples to `adcDmaBuffer[x]`), the `TIMER2_IRQHandler` **must** be called and run up to `END_CTRITICAL_SECTION` **within** one ADC interval (`100us`).

Most of the time this works correctly, how ever, sometimes we saw that SPIM3 transfered 105 samples before we could swap the buffers!

```c
static const nrfx_spim_t spi3 = NRFX_SPIM_INSTANCE(3);
static nrf_ppi_channel_t ppi_Adc  = NRF_PPI_CHANNEL31;

static nrf_ppi_channel_t ppiChannel_AdcTimerCounter  = NRF_PPI_CHANNEL31;
static const nrfx_timer_t conv_counter = NRFX_TIMER_INSTANCE(2);

uint8_t adcDmaBuffer[2][30][ADC_SAMPLE_SIZE]; // ADC DMA Buffers (Dualbuffer with 2 slots)
uint8_t adcDmaSlotIndex = 0;

void init(void) {
    /* Timer Counter init */
    NRF_TIMER2->MODE = TIMER_MODE_MODE_Counter;             // Set the timer in Counter Mode
    NRF_TIMER2->TASKS_CLEAR = 1;                            // clear the task first to be usable for later
    NRF_TIMER2->BITMODE = TIMER_BITMODE_BITMODE_16Bit;      // Set counter to 16 bit resolution
    NRF_TIMER2->CC[0] = ADC_DMA_TRANSFER_SIZE_SAMPLES;      // Set value for TIMER2 compare register 0 (Number of samples to fit in the DMA buffer)
    
    // Enable interrupt on Timer 2 compare match events
    NRF_TIMER2->INTENSET = TIMER_INTENSET_COMPARE0_Enabled << TIMER_INTENSET_COMPARE0_Pos;
    
    // Enable shortcut on Compare Event
    NRF_TIMER2->SHORTS = TIMER_SHORTS_COMPARE0_CLEAR_Enabled << TIMER_SHORTS_COMPARE0_CLEAR_Pos;
    
    IRQ_DIRECT_CONNECT(TIMER2_IRQn, TIMER_IRQ_PRIORITY, TIMER2_IRQHandler, 0);
    
    nrfx_ppi_channel_alloc(&ppiChannel_AdcTimerCounter); // Connect ADC_DRDY falling edge event to counter input event
    nrfx_ppi_channel_assign(ppiChannel_AdcTimerCounter, nrfx_gpiote_in_event_addr_get(ADC_nDRDY), nrfx_timer_task_address_get(&conv_counter, NRF_TIMER_TASK_COUNT));
    
    NVIC_EnableIRQ(TIMER2_IRQn); // Enable the interrupt

    
    /* ADC Init */
    NRF_SPIM3->RXD.MAXCNT = ADC_SAMPLE_SIZE;
    NRF_SPIM3->RXD.PTR = (uint32_t) &adcDmaBuffer[adcDmaSlotIndex];
    NRF_SPIM3->RXD.LIST = 1;
    
    nrfx_gpiote_in_event_enable(ADC_nDRDY, true);
    nrfx_ppi_channel_alloc(&ppi_Adc);
    nrfx_ppi_channel_assign(ppi_Adc, nrfx_gpiote_in_event_addr_get(ADC_nDRDY), nrfx_spim_start_task_get(&spi3);
    nrfx_ppi_channel_enable(ppi_Adc);
}

/* Timer Handler */
ISR_DIRECT_DECLARE(TIMER2_IRQHandler) {
    uint32_t dmaWritePtr;
    NRF_TIMER2->EVENTS_COMPARE[0] = 0; // Clear compare register 0 event
    dmaWritePtr = NRF_SPIM3->RXD.PTR; // Backup for later checks (how many samples did effectively got written until we got called)

    if(adcDmaSlotIndex == ADC_DMA_SLOT_0){
        NRF_SPIM3->RXD.PTR = (uint32_t) &adcDmaBuffer[ADC_DMA_SLOT_1];
        adcDmaSlotIndex = ADC_DMA_SLOT_1;
    }
    else {
        NRF_SPIM3->RXD.PTR = (uint32_t) &adcDmaBuffer[ADC_DMA_SLOT_0];
        adcDmaSlotIndex = ADC_DMA_SLOT_0;
    }

    // END_CTRITICAL_SECTION

    /* From now on we have time to do what ever we need. */

    /* Check how many samples effectively got copied by SPIM3 */
    adcDmaSlotIndex_t adcDmaSlotPreviousIndex;
    /* The slots got already swapped, but we need to access the now inactive slot */
    if(adcDmaSlotIndex == ADC_DMA_SLOT_0){
        adcDmaSlotPreviousIndex = ADC_DMA_SLOT_1;
    }
    else {
        adcDmaSlotPreviousIndex = ADC_DMA_SLOT_0;
    }
    dmaBytesTranfered = dmaWritePtr - (uint32_t)&adcDmaBuffer[adcDmaSlotPreviousIndex]; // subtract buffer start by buffer start address => transfered bytes
    samplesInDmaBuffer = dmaBytesTranfered / ADC_SAMPLE_SIZE;

    __ASSERT(samplesInDmaBuffer > 30, "We were too slow, the SPIM3 transfered %d samples instead of only 30! => RAM corruption!!!", samplesInDmaBuffer);

    [..]

    return 1;
}

```

caco3, 2021-12-21:
Update question:
Would it somehow be possible to update the `NRF_SPIM3->RXD.PTR` by PPI? That would solve the whole issue.

carlescufi, 2021-12-21:
> Update question: Would it somehow be possible to update the `NRF_SPIM3->RXD.PTR` by PPI (That is, RAM to RAM transfer)? That would solve the whole issue.

Maybe @anangl can answer that

cvinayak, 2021-12-21:
What value are you using for `TIMER_IRQ_PRIORITY`?

caco3, 2021-12-21:
> What value are you using for `TIMER_IRQ_PRIORITY`?

I used priority 1.
Below a table of some notes (based on your explanations above):

[image]

Would it work to change my timer interrupt priority also to 0?

cvinayak, 2021-12-21:
Ok, continue to keep `TIMER_IRQ_PRIORITY=1`.

Did you have these?
`CONFIG_BT_CTLR_ADVANCED_FEATURES=y` and
`CONFIG_BT_CTLR_ULL_HIGH_PRIO=2`

And please use lower priorities for other SoC peripherals used in your application by having the below

`#define NRF_DEFAULT_IRQ_PRIORITY 5`

inside your application folder in `dts/arm/nordic/override.dtsi`  file.

Maybe 10.5 ms latency seen could be from other SoC Peripheral ISRs which are by default at value 1.

caco3, 2021-12-21:
> Ok, continue to keep `TIMER_IRQ_PRIORITY=1`.

ok

> Did you have these? `CONFIG_BT_CTLR_ADVANCED_FEATURES=y` and `CONFIG_BT_CTLR_ULL_HIGH_PRIO=2`

2x yes

> And please use lower priorities for other SoC peripherals used in your application by having the below
> `#define NRF_DEFAULT_IRQ_PRIORITY 5`
> inside your application folder in [...]

Is there a way to check if this got correctly applied?
I have my own overlay and dts files so I would like to check this at runtime.

I will have to let it run for a few hours to check if it helps.

---
Something else I discovered, maybe not directly related to the IRQ priorities:
I often see (and also already saw before the `NRF_DEFAULT_IRQ_PRIORITY 5` change),  that the SPIM3 had 29 or 31 samples transfered at the time the `TIMER2_IRQHandler` gets called:
```
[00:04:10.565,246] <dbg> ads131.: dmaWritePtr: 0x200144E0, (Slot 1 start: 0x200143AA)
[00:04:10.565,246] <dbg> ads131.: DMA transfered 310 bytes (31 samples) but should have 300 (IRQ Count: 70440, slot: 1)!

[00:04:10.568,145] <dbg> ads131.: dmaWritePtr: 0x2001430A,  (Slot 0 start: 0x200141E8)
[00:04:10.568,176] <dbg> ads131.: DMA transfered 290 bytes (29 samples) but should have 300 (IRQ Count: 70441, slot: 0)!
```
I think this might happen when `TIMER2_IRQHandler` once gets called with a latency between `100..200us` , SPIM3 at that time already transfered 31 samples, and at the next time Timer2 fires already when SPIM3 has only 29 samples transfered because the timer counter still correctly wrapped at 30. **This is something I can handle fine.**

How ever, sporadically I see that the `TIMER2_IRQHandler` fires but SPIM3 has **only 15 samples** transfered:
```
[00:06:51.201,751] <dbg> ads131.: dmaWritePtr: 0x20014440, 0x200141E8, 0x200143AA
[00:06:51.201,751] <dbg> ads131.: DMA transfered 150 bytes (15 samples) but should have 300 (IRQ Count: 123984, slot: 1)! This should never happen!
```
This doesn't make any sense to me, especially because before and after this call it is correctly 30.
Since the SPIM3 and Timer2 have the same event EEP, why can one get a different number of calls than the other?
I also once replaced the Timer2 EEP with a `SPIM3 Transfer END` event, but no change:
```
ret = nrfx_ppi_channel_assign(ppiChannel_AdcTimerCounter, nrfx_spim_end_event_get(&spi3), nrfx_timer_task_address_get(&conv_counter, NRF_TIMER_TASK_COUNT));
```

cvinayak, 2021-12-21:
> Is there a way to check if this got correctly applied?
I have my own overlay and dts files so I would like to check this at runtime.

You can check the `zephyr.dts` file, the interrupt priorities should be 5 instead of 1.