zephyrproject-rtos/zephyr #41281: Style Requirements Seem to Be Inconsistent with Uncrustify Configuration

catch-twenty-two, 2021-12-17:
There seem to be an inconsistency with the required stylistic formatting outlined in the style guide and the formatting settings for Uncrustify. This makes it difficult for new code submissions to conform to what is outlined in the contribution guide and what is expected from the community. 

> > Looks good to me. I'd honestly suggest filing an issue on uncrustify and the docs here if you ran it and it still didn't conform. The docs suggest using it, I've done just that as well, and yet folks still note that it doesn't conform to the style guide which is I agree a confusing thing to run into. Especially as a new contributor.
> 
> Agreed, we are preparing to submit more drivers and updates to drivers, I'd hate for things to blocked because of stylistic issues that can't be address without multiple updates to prs.

I agree that it's annoying, and unfortunately tooling like uncrustify seem to be outdated or not working as expected (there are some compliance issues that are likely related to that). You may try with clang-format, from my experience it is not perfect but does a decent job.

_Originally posted by @gmarull in https://github.com/zephyrproject-rtos/zephyr/issues/37749#issuecomment-992269364_

lairdjm, 2022-04-11:
This also includes the clang-format file. An example, file contains:
```
const struct hash_checksum_mgmt_group *hash_checksum_mgmt_find_handler(const char *name)
{
        /* Find the group with the specified group name */
        struct hash_checksum_mgmt_group *group = NULL;
        sys_snode_t *snp, *sns;

        SYS_SLIST_FOR_EACH_NODE_SAFE(&hash_checksum_mgmt_group_list, snp, sns) {
                struct hash_checksum_mgmt_group *loop_group =
```

Note that there is no space between `SYS_SLIST_FOR_EACH_NODE_SAFE` and the bracket, however, running this file through clang-format with the zephyr `.clang-format` file puts a space between them, then the PR fails because there is a space where there should not be a space.

gmarull, 2022-04-16:
@lairdjm FYI https://github.com/zephyrproject-rtos/zephyr/pull/41307 (I'll pick this again when I come back from paternity leave in a few weeks)

gmarull, 2022-06-16:
We have updated clang-format and removed uncrustify in https://github.com/zephyrproject-rtos/zephyr/pull/41307

closed 2022-06-16 by gmarull