zephyrproject-rtos/zephyr #41286: Bluetooth SDP: When the SDP attribute length is greater than SDP_MTU, the attribute is discarded

LuoZhongYao, 2021-12-17:
**Describe the bug**
When the length of an attribute in the SDP record is greater than SDP_MTU,
the select_attrs function discards the attribute.
 - ...

**To Reproduce**
1. Configure an SDP service so that the length of one of the attributes is greater than SDP_MTU
2. Connect ZEPHYR via Bluetooth
3. Through the Bluetooth HCI log, it can be found that the attribute in the SDP returned by ZEPHYR
    returned an empty packet 

**Expected behavior**
SDP should divide the attribute into pieces and return the complete SDP record through multiple
BT_SDP_SVC_ATTR_REQ instructions 

**Impact**
HID report descriptor cannot be reported

dleach02, 2022-01-04:
@MarkWangChinese can you look at this?

MarkWangChinese, 2022-01-05:
Hi @LuoZhongYao  do you mean that the one HID report descriptor attribute's data length is bigger than the SDP_MTU?

LuoZhongYao, 2022-01-06:
@MarkWangChinese Any attribute greater than SDP_MTU will do, but the HID report descriptor is usually larger, and it is easier to exceed SDP_MTU

MarkWangChinese, 2022-01-26:
Hi @LuoZhongYao  please have a try with https://github.com/zephyrproject-rtos/zephyr/pull/42166

dleach02, 2022-04-08:
@louzhongyao, do you have any feedback to @MarkWangChinese comment?

LuoZhongYao, 2022-04-11:
@dleach02 @MarkWangChinese  Can each request return MTU bytes instead of an attribute

MarkWangChinese, 2022-04-15:
> @dleach02 @MarkWangChinese Can each request return MTU bytes instead of an attribute

@LuoZhongYao  I don't see big benefit to return MTU bytes for every transfer except the throughput is improved. But the SDP don't need run many times for one connection, there is no big throughput requirement. I want to put the fixed codes to one MACRO to reduce the code/ram for most user cases that's ATT length is not bigger than SDP_MTU. If the MACRO is enabled, the more code/ram is required. If I am wrong, please let me know.

LuoZhongYao, 2022-04-15:
@MarkWangChinese I verified that this RP is valid

closed 2022-07-30 by github-actions[bot]