zephyrproject-rtos/zephyr #41291: LVGL touch event "LV_EVENT_LONG_PRESSED" can not be generated if I press the screen without lift up my finger

glin00, 2021-12-17:
**Describe the bug**
I am using lvgl as GUI library with zephyr.  The display which is an OLED display can be displayed ok using lvgl.  The OLED display touch function is working since I touch the display, the correct display x,y can be reported.  If I press the screen and lift up my finger, "LV_EVENT_CLICKED" can be generated within lvgl. But if I press the screen without lift up my finger ,"LV_EVENT_LONG_PRESSED" can not be generated.    
. 

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

glin00, 2021-12-20:
After I trace code, I found that ,within indev_proc_press function (line 975), proc->pr_timestamp (press screen time stamp) always equal to sys_time.  This causes lv_tick_elaps return a value not greater than indev_act->driver.long_press_time. 
 As a result, LV_SIGNAL_LONG_PRESS can not be triggered. 
Can any one help why  proc->pr_timestamp is always equal to sys_time?

carlescufi, 2021-12-21:
@gmarull or @brgl any ideas?

gmarull, 2021-12-23:
`proc.pr_timestamp` seems to be updated in some places using `lv_tick_get()`, which is a shim for `k_uptime_get_32()`, so the problem may come from input events generated by the corresponding kscan driver. Check lvgl_pointer_kscan_read and related functions.

dkalowsk, 2022-01-04:
@kisvegabor any input?

kisvegabor, 2022-01-05:
I've answered in the same in issue in https://github.com/lvgl/lvgl/issues/2923 but didn't get answer :slightly_frowning_face:

closed 2022-01-07 by glin00