zephyrproject-rtos/zephyr #41297: QSPI flash need read, write via 4 lines not 1 line

glin00, 2021-12-17:
**Describe the bug**
A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

erwango, 2021-12-17:
No issue description. Closing

glin00, 2021-12-17:
@erwango , please open up this issue since I had filled up all the information I know. If there is any information not enough, please let me know. Thanks.

erwango, 2021-12-17:
@glin00 Can you have a look to https://github.com/zephyrproject-rtos/zephyr/pull/40859, to see if filling the gap ?

erwango, 2021-12-17:
Also, @glin00 please make the effort to distinguish 'bug' from 'enhacement'.

lmajewski, 2021-12-20:
@glin00 The PR #40859 mentioned by @erwango shall fulfill this enhancement request.

glin00, 2022-02-10:
@lmajewski . would you mind to tell me the SHA1 ID of zephyr you are using to test #40859?
I am using SHA1 ID b48276e9954a11ccf1ebeb93c635f5f16fe8fdb2 and I copy paste the files you changed in #40859 and there are some compiling errors which I am trying to solve now.
But I think the quickest way is to know which SHA1 ID of zephyr you are using, so I can spend less time to solve this problem.

lmajewski, 2022-02-10:
@glin00 

Compiler: Found host-tools: zephyr 0.13.1
Branch: upstream/main
SHA1: 9e4968bb901c8dddde661b8f2d0a0046b338a8ab
Compilation: west build -p always -b nucleo_h743zi ./zephyr/samples/subsys/fs/littlefs
Flashing: west flash

lmajewski, 2022-02-10:
IIRC there was a pinctrl rework:

`tig b48276e9954a11ccf1ebeb93c635f5f16fe8fdb2..9e4968bb901c8dddde661b8f2d0a0046b338a8ab`

Please look into: SHA1: 37bf7cbe4f3bd7239b6675e9f9b3e7746fde193f

erwango, 2022-03-21:
Fixed in https://github.com/zephyrproject-rtos/zephyr/pull/40859,.
Please open a new issue if required

closed 2022-03-21 by erwango