zephyrproject-rtos/zephyr #41299: IS25LP016D SPI NOR FLASH PROBLEM

mmterkc, 2021-12-17:
Hi everyone, I use custom board. Custom board have nrf52833 and IS25LP016D SPI NOR FLASH. I changed device tree for my custom board but I get SPI flash driver is25lp016d was not found! error. How can I solve these problem? 
My code:
/*
 * Copyright (c) 2016 Intel Corporation.
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <zephyr.h>
#include <drivers/flash.h>
#include <device.h>
#include <devicetree.h>
#include <stdio.h>
#include <string.h>
#include <drivers/gpio.h>

#if (CONFIG_SPI_NOR - 0) ||				\
	DT_NODE_HAS_STATUS(DT_INST(0, jedec_spi_nor), okay)
#define FLASH_DEVICE DT_LABEL(DT_INST(0, jedec_spi_nor))
#define FLASH_NAME "JEDEC SPI-NOR"
#elif (CONFIG_NORDIC_QSPI_NOR - 0) || \
	DT_NODE_HAS_STATUS(DT_INST(0, nordic_qspi_nor), okay)
#define FLASH_DEVICE DT_LABEL(DT_INST(0, nordic_qspi_nor))
#define FLASH_NAME "JEDEC QSPI-NOR"
#elif DT_NODE_HAS_STATUS(DT_INST(0, st_stm32_qspi_nor), okay)
#define FLASH_DEVICE DT_LABEL(DT_INST(0, st_stm32_qspi_nor))
#define FLASH_NAME "JEDEC QSPI-NOR"
#else
#error Unsupported flash driver
#endif

#if defined(CONFIG_BOARD_ADAFRUIT_FEATHER_STM32F405)
#define FLASH_TEST_REGION_OFFSET 0xf000
#elif defined(CONFIG_BOARD_ARTY_A7_ARM_DESIGNSTART_M1) || \
	defined(CONFIG_BOARD_ARTY_A7_ARM_DESIGNSTART_M3)
/* The FPGA bitstream is stored in the lower 536 sectors of the flash. */
#define FLASH_TEST_REGION_OFFSET \
	DT_REG_SIZE(DT_NODE_BY_FIXED_PARTITION_LABEL(fpga_bitstream))
#else
#define FLASH_TEST_REGION_OFFSET 0xff000
#endif
#define FLASH_SECTOR_SIZE        4096

void main(void)
{	const struct device *gpio0;
	const struct device *gpio1;
	gpio0 = device_get_binding("GPIO_0");
	gpio1 = device_get_binding("GPIO_1");
	gpio_pin_configure(gpio0,4,GPIO_OUTPUT);
	
	
	const uint8_t expected[] = { 0x55, 0xaa, 0x66, 0x99 };
	const size_t len = sizeof(expected);
	uint8_t buf[sizeof(expected)];
	const struct device *flash_dev;
	int rc;

	printf("\n" FLASH_NAME " SPI flash testing\n");
	printf("==========================\n");

	flash_dev = device_get_binding(FLASH_DEVICE);

	if (!flash_dev) {
		printf(FLASH_DEVICE);
		printf("SPI flash driver %s was not found!\n",
		       FLASH_DEVICE);
		return;
	}

	/* Write protection needs to be disabled before each write or
[... 537 lines cut ...]
};

My proj.conf:
CONFIG_STDOUT_CONSOLE=y
CONFIG_SPI=y

CONFIG_SPI_NOR=y
CONFIG_SPI_NOR_FLASH_LAYOUT_PAGE_SIZE=4096
CONFIG_GPIO=y
CONFIG_SEGGER_SYSTEMVIEW=n
CONFIG_LOG=y
CONFIG_FLASH_LOG_LEVEL_DBG=y

my custom board pins:

[image]
[image]
[image]
[image]
[image]

si=mosi
so=miso
ce=chip-enable
sck=clock

nvlsianpu, 2022-01-05:
Did you tried limit the device name to `#define FLASH_DEVICE DT_LABEL(DT_INST(0, jedec_spi_nor))`
Also: IS25LP: IS25LP016D@0 node compatible should be probably limited to `"jedec,spi-nor"`

carlescufi, 2022-02-01:
Closing due to inactivity. If this is still an issue please reopen.

closed 2022-02-01 by carlescufi