zephyrproject-rtos/zephyr #41313: LwM2M: make transport agnostic

rerickson1, 2021-12-17:
**Is your enhancement proposal related to a problem? Please describe.**
Currently the LwM2M engine is tightly coupled to sending/receiving data with sockets.
There's no reason this can't be decouple to allow sending over other transports therefore removing the dependency on the networking stack.

**Describe the solution you'd like**
Decouple the LwM2M engine from networking and sockets.
1. Decouple the engine from sockets and add a Kconfig option to use the socket implementation by default
2. If Kconfig option for using sockets is turned off, callbacks for LwM2M send/receive can be implemented by the user app.

This would allow a user to tunnel LwM2M over a BLE connection for instance.

rerickson1, 2021-12-17:
@rlubos curious if you have any thoughts or further input on this.

rlubos, 2021-12-20:
We have something like this in MQTT, where we implement multiple transports, and user can select which one to use when initializing client (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/mqtt/mqtt_transport.h) - I think we could implement something similar for LwM2M as well. 

Or were you rather thinking about being able to disable LwM2M thread entirely, and handle the communication on the application side? The LwM2M would then need an API to feed the data into the stack and to trigger send.

rerickson1, 2021-12-22:
Thanks for pointing out that feature of MQTT, I was not aware of it. I think I would like to start with an approach similar to that for LwM2M.

0Grit, 2021-12-22:
Recommend reading the LWM2M transport spec

http://www.openmobilealliance.org/release/LightweightM2M/V1_2-20201110-A/OMA-TS-LightweightM2M_Transport-V1_2-20201110-A.pdf

rerickson1, 2021-12-22:
@ahedin25