zephyrproject-rtos/zephyr #41317: ADSP: Many kernel test cases which have CONFIG_MP_NUM_CPUS=1 failed in daily testing

enjiamai, 2021-12-18:
**Describe the bug**
About one half of test cases failed. The symptom is it hangs up without any log output. 
It only happens on test cases which CONFIG_MP_NUM_CPUS=1

 - What target platform are you using?  intel_adsp_cavs15
 - What have you tried to diagnose or workaround this issue? The PR https://github.com/zephyrproject-rtos/zephyr/pull/41283 will fix this problem. 

**To Reproduce**
Steps to reproduce the behavior:
This is one example of running tests/kernel/mutex/mutex_api, any test case which has CONFIG_MP_NUM_CPUS=1 could reproduce it.
1. source zephry-env.sh
2. twister -p intel_adsp_cavs15 --device-testing -T tests/kernel/mutex/mutex_api --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh,/home/ztest/work/zephyrproject/modules/audio/sof/keys/otc_private_key.pem,/home/ztest/work/rimage/config,/home/ztest/work/rimage/build/rimage" --device-serial-pty="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py,--no-history" -vv

**Expected behavior**
The test case pass and show up its test result.

**Impact**
Many test cases failed, the pass rate drop about half.

**Logs and console output**
```
===
=== [Invalid slot 0; ADSP Device Reset?]
===
```
Then it will hang up until timeout, test cases failed.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain e.g Zephyr SDK 0.13.4
 - Commit SHA or Version used:

andyross, 2022-01-18:
Is this resolved?  Can we retest and update?

enjiamai, 2022-01-19:
Hi @andyross , I am verifying this in ADSP, TGL... please wait me for couple hours, feedback you soon. thanks.

enjiamai, 2022-01-20:
Verified this can be closed after #41283 merged.

closed 2022-01-20 by enjiamai