zephyrproject-rtos/zephyr #41318: stm32: counter: making counter date-time friendly

manoj153, 2021-12-18:
I have great interest to support the RTC datetime feature for STM32 mainly.

Should I poke the counter driver and its subsystem?

It is a simple exercise to introduce 2 API as a solution:

- set_system_time(struct tm *now)
- get_system_time(struct tm *now)

Please let me know if this is the right way of doing it?

ycsin, 2021-12-20:
You might be interested in #23526

erwango, 2021-12-20:
I think the recent RTC API proposal would answer your request: https://github.com/zephyrproject-rtos/zephyr/pull/41287.
Question remains: how to make it works with https://github.com/zephyrproject-rtos/zephyr/pull/28977

erwango, 2023-04-05:
> I think this is covered by https://github.com/zephyrproject-rtos/zephyr/issues/35333.

Ok, closing current. Thanks

closed 2023-04-05 by erwango