zephyrproject-rtos/zephyr #41331: tests: drivers: disk: fail to handle no SD card situation

LixinGuoX, 2021-12-20:
**Describe the bug**
When run this test case on boards with no SD cards inserted, test gets failed. I think this case should be
skipped if the hardware does not match the requirement. Therefore, if you run many test cases together,
you will not get unexpected failure.

**To Reproduce**
Steps to reproduce the behavior:
run test cases on board with no SD card inserted.

**Expected behavior**
Boards with no SD card inserted should skip this test instead of getting failed.

**Logs and console output**
```
*** Booting Zephyr OS build v2.7.99-2572-gfef2e30b7f83  ***
Running test suite disk_driver_test
===================================================================
START - test_setup
E: NO SD inserted!
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/disk/src/main.c:50: test_setup: (rc not equal to 0)
Disk access initialization failed
FAIL - test_setup in 0.115 seconds
===================================================================
START - test_read
Testing reads of 8 sectors
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/disk/src/main.c:95: test_sector_read: (rc not equal to 0)
Failed to read from sector zero
FAIL - test_read in 0.16 seconds
===================================================================
START - test_write
Testing writes of 8 sectors
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/disk/src/main.c:148: test_sector_write: (rc not equal to 0)
Failed to write to sector zero
FAIL - test_write in 0.16 seconds
===================================================================
Test suite disk_driver_test failed.
===================================================================
PROJECT EXECUTION FAILED
```

danieldegrasse, 2022-01-05:
@LixinGuoX Won't other tests, like the [SPI loopback example](https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/drivers/spi/spi_loopback), fail if the device under test does not have a specific hardware configuration (jumper between MISO and MOSI)? Or is there a method you're using to exclude those tests that the disk driver test does not support? If so, maybe that could be added here, to enable the user to skip this test when running a large test suite.

My concern is that skipping the test if an SD card was not inserted would break the generic nature of the test, since the [disk access api](https://docs.zephyrproject.org/latest/reference/storage/disk/access.html) this tests against does not have a defined return code for disk media not being present. If a check was added, then this test would become linked with the USDHC driver, rather than other disk drivers present in Zephyr

jfischer-no, 2022-01-05:
@danieldegrasse I guess the testcase.yaml needs "fixture: disk_sdcard", like there is`fixture: spi_loopback` or `fixture: gpio_loopback`.

danieldegrasse, 2022-01-06:
@jfischer-no Thank you for the idea, that seems like it should work. @LixinGuoX, could you look at #41629 and confirm it works for your test case?

LixinGuoX, 2022-01-07:
@danieldegrasse , yes, it works, now test is skipped instead of getting failed. Thanks.

PR #41629 (2022-01-07): tests: disk: Add fixture for sdcard test
Add fixture disk_sdcard for the sdcard test, since the test requires an
SD card to be present on the board.

Fixes #41331
commit e6cac6f0 (2022-01-07): tests: disk: Add fixture for sdcard test

Add fixture disk_sdcard for the sdcard test, since the test requires an
SD card to be present on the board.

Fixes #41331
closed 2022-01-07 by MaureenHelm