zephyrproject-rtos/zephyr #41339: stm32, Unable to read UART while checking from Framing error.

andreacampanella, 2021-12-20:
I'm implementing DMX512 input on an F4.

Reading it's done at 250000 bauds, the Break and the Mark After Break it's at 100000 so it's common practice to detect a framing error instead of switching baudrate or edge detection. 

I have an interrupt handler for the UART IRQ, inside it I have this piece of code: 

    while (uart_irq_update(dev) && uart_irq_is_pending(dev)) {
       if (uart_irq_rx_ready(dev)) {   
            int err = uart_err_check(dev);
            if (err > 0)
            {
                if(err == UART_ERROR_OVERRUN ){
                    printk("[UART0_RXTX] UART_ERROR_OVERRUN \n");
                }
                if(err == UART_ERROR_PARITY  ){
                    printk("[UART0_RXTX] UART_ERROR_PARITY \n ");
                }
                if(err == UART_ERROR_FRAMING ){
                    printk("[UART0_RXTX] UART_ERROR_FRAMING \n ");
                }
                if(err == UART_BREAK ){
                    printk("[UART0_RXTX] UART_BREAK \n");
                }
                if(err == UART_ERROR_COLLISION ){
                    printk("[UART0_RXTX] UART_ERROR_COLLISION \n");
                }
            }
            int ret = uart_fifo_read(dev,&_rx_buffer,1);
            if (ret > 0)
            {
                printk("read data %d\n"_rx_buffer);
            }
            if (ret < 0)
            {
                printk("error reading data\n");
            }
            else{
                printk("no data\n");   
            }
        }

I should be able to detect the Break or the data but if I detect the Break , uart_fifo_read always returns 0,  this doesn't happen if I try to not detect the break.

I'm expecting to be able to check the error flags and be able to read from the buffer if no error has occured.

erwango, 2021-12-21:
Seems that Line Break Detection is missing from `uart_stm32_err_check`;
I'm not sure about the reason why it is missing but we can try to fix this.

I can't test this myself right now, but maybe you can have a try to add the following, in `drivers/uart_stm32.c`, in function ``, just after the framing error detection and before clearing errors ([here](https://github.com/zephyrproject-rtos/zephyr/blob/682f261c95ef8fbbd9f587e36f466f9382611454/drivers/serial/uart_stm32.c#L592)):
```
	if (LL_USART_IsActiveFlag_LBD(UartInstance)) {
		err |= UART_BREAK;
	}

	if (err & UART_BREAK) {
		LL_USART_ClearFlag_LBD(UartInstance);
	}
```

Can you try this and see if fixing your point ?

Please note that by default, line break detection is set for 10-bit break. You'll need to make some changes for 11-bit break detection.

andreacampanella, 2021-12-21:
Hey thanks about your answer, the modification seems to detect Breaks but I would like to test if futher, I will update when I have more data. 

Do you want me to do a PR in case?

erwango, 2021-12-21:
> Hey thanks about your answer, the modification seems to detect Breaks but I would like to test if futher, I will update when I have more data.
> 
> Do you want me to do a PR in case?

Thanks for the tests. Yes, don't hesitate to post a PR once you believe this is good to merge.

andreacampanella, 2021-12-21:
I kept an eye on it all day and it's now behaved as expected, I created a PR.

FRASTM, 2022-01-19:
@andreacampanella  can you please check comment https://github.com/zephyrproject-rtos/zephyr/issues/41818#issuecomment-1016218393  to see if it is linked to your pb ?

closed 2022-01-28 by carlescufi