zephyrproject-rtos/zephyr #41350: Why use net_if_get_first_by_type(...) instead of net_if_get_default in dhcp4 test case from v.2.6.0.

kevinwang821020, 2021-12-21:
In my opinion, is using net_if_get_default more appropriate than using net_if_get_first_by_type(&NET_L2_GET_NAME(DUMMY)) in tests/net/dhcpv4/src/main.c:434?

Reason:
If using the net_if_get_first_by_type(&NET_L2_GET_NAME(DUMMY)), the test program can only be tested through a fake network interface(GUMMY), it means that it won't  test the communicate between platform and DHCP server on the real network. But if using net_if_get_default, users can use Kconfig to select the default network interface, and if the developer has ported the driver for physical network interface, users can test the communication between the device and DHCP server.

Maybe it’s because I don’t know enough about the program architecture
Please forgive me if the concept is not correct.

Thanks.

jukkar, 2021-12-21:
Commented this question on mailing already.

The unit test programs are self contained i.e., they do not need any external network connections to verify functionality. If the unit test is run in a board that has for example Ethernet support, there could be multiple network interfaces created (Ethernet and Dummy). If the unit test would use net_if_get_default(), then we might get incorrect network interface returned. So to avoid this possible issue, the unit tests use the proper network interface in order to allow the test pass.

jukkar, 2021-12-21:
Converting this to discussion as this is more appropriate there.