zephyrproject-rtos/zephyr #41354: Linker: ARM64: the .bss section is included in zephyr.bin file.

Nam97UTC, 2021-12-21:
Hello everyone, I have a concern about linker file of ARM64 as described below:
### Description
With the `.bss` section is placed above `.data`, so binary file always constains `.bss` section.
https://github.com/zephyrproject-rtos/zephyr/blob/38ce8d615d4b9d97e1b86d400a2c90ab75bbb5bd/include/arch/arm64/scripts/linker.ld#L237-L291
I saw that the linker file of ARM cortex-M7.

- If `CONFIG_USERSPACE = y` --> file bin contains `.bss` section

- If `CONFIG_USERSPACE = n`--> file bin does not contain `.bss` section

https://github.com/zephyrproject-rtos/zephyr/blob/38ce8d615d4b9d97e1b86d400a2c90ab75bbb5bd/include/arch/arm/aarch32/cortex_m/scripts/linker.ld#L256-L377
### Concerns
I don't know why the ARM64 always put `.bss` section there. Will there be any effect if I use `CONFIG_USERSPACE ` to put `.bss` section like the linker ARM cortex-M?
**Can someone explain it help me. Thanks in advance.**

carlocaione, 2021-12-21:
> * If `CONFIG_USERSPACE = n`--> file bin does not contain `.bss` section

It still does:

https://github.com/zephyrproject-rtos/zephyr/blob/main/include/arch/arm/aarch32/cortex_m/scripts/linker.ld#L333-L358

just at a different offset.

tejlmand, 2021-12-21:
@Nam97UTC  please read the history of this PR https://github.com/zephyrproject-rtos/zephyr/pull/21747 as to why there is this difference on the arm cortex-M linker script.

Nam97UTC, 2021-12-21:
Oki, thank you so much, I found it.

closed 2021-12-21 by Nam97UTC