zephyrproject-rtos/zephyr #41359: Bluetooth: connection times when trying to connect from certain centrals

jsiverskog, 2021-12-21:
**Describe the bug**
we're having a system consisting of (among other things) an nrf52840, running zephyr. it runs bluetooth (le) both as central and peripheral.

when acting as peripheral (no other connection set up), we're having issues when connecting from certain centrals. shortly after connecting we receive a disconnected event with reason 8 (timed out). to connect, we're using "audio midi setup" in macos (this is a ble midi device).

the issue occurs with at least two different centrals:
- macbook pro late 2016, running macos 10.15.6
- imac retina 5k 27" 2017, running macos 11.6

it *does* work from other macs, for instance macbook air mid 2012 running macos 10.15.7.

**To Reproduce**
Steps to reproduce the behavior:
1. flash system
2. enable advertising
3. connect with one of these centrals

**Expected behavior**
connection stable.

**Impact**
showstopper.

**Logs and console output**
[sniff.txt](https://github.com/zephyrproject-rtos/zephyr/files/7751249/sniff2.txt) (rtt logs from a number of connect/disconnect events)
[capture.zip](https://github.com/zephyrproject-rtos/zephyr/files/7751244/capture.zip) (wireshark capture from one of these)
[config.txt](https://github.com/zephyrproject-rtos/zephyr/files/7751287/config.txt) (.config from build)

**Environment (please complete the following information):**
 - OS: macos
 - Toolchain: gcc-arm-none-eabi-10.3-2021.07
 - 2.7.0, 2.7.1, fef2e30b7f8346c28e84b667b05b251133823676 (zephyr head yesterday)

**Additional context**
we believe this is a regression and that it has been working before, on 2.5 or 2.6. unfortunately it's difficult to get logs/wireshark captures from those versions due to reasons.

cvinayak, 2021-12-21:
From the sniffer logs:
The central is a Bluetooth v4.2 device:
[image]

Central performs invalid procedure sequence, central should send PHY_UPDATE_IND or Unknown PDU response, not PHY_RSP
[image]

The Central being  a v4.2 implementation, shall be responding with Unknown PDU response, this is a violation from the central implementation.

Screenshot for PHY update procedure MSC from Bluetooth Specification v5.3:
[image]

Quirks/Solutions:
1. Zephyr application workaround is to not use auto PHY update feature, and use the user initiated PHY update after checking the peer Bluetooth Version.
2. Update controller to fail PHY update if feature exchange was not done prior or remoted supported features response does not have PHY update support. This would be a controller design change needing re-run of conformance tests, and could be non-compliant if controller does not send PHY_REQ to centrals that do not support the feature as part of a test, if any.

Recommended action:
- Application to implement User Initiated PHY Update (have `CONFIG_BT_USER_PHY_UPDATE=y`, and `CONFIG_BT_AUTO_PHY_UPDATE=n`)

cvinayak, 2021-12-21:
@jsiverskog Could this help? https://github.com/zephyrproject-rtos/zephyr/blob/8b3d7f2a4368e0144f2b9d57f9939cd438604b61/include/bluetooth/conn.h#L420-L422

You could initiate the PHY update in/after this callback `remote_info_available` after a check on the version?

carlescufi, 2021-12-21:
@jsiverskog I think @joerchan might be able to help with this here or on Discord.

jsiverskog, 2021-12-21:
@cvinayak : that's what i tried, but the `remote_info_available` callback is called _after_ remote features have been read out, and according to the spec we can't trigger a phy update after that.

@carlescufi : i'll hit him up on discord. thanks.

cvinayak, 2021-12-21:
>  the `remote_info_available` callback is called _after_ remote features have been read out, and according to the spec we can't trigger a phy update after that.

Ok, may be I need to fix the controller then, to not perform the host initiated auto PHY update and give back a feature not supported error back.

Hmm, its already implemented that way, if PHY update is requested and controller has seen a feature exchange then the controller will return error.... see here: https://github.com/zephyrproject-rtos/zephyr/blob/8b3d7f2a4368e0144f2b9d57f9939cd438604b61/subsys/bluetooth/controller/ll_sw/ull_conn.c#L797-L803

Hence, if you do initiate a user PHY update after `remote_info_available` then controller should do the right thing and decide to not initiate PHY update if the feature is not supported by the peer device.

jsiverskog, 2021-12-21:
@cvinayak : i think you're right. but perhaps i'm running into some other issue. when i sniff i see the `LL_PHY_REQ` going out, but the central replies with `LL_REJECT_IND_EXT` (with error code 0x2a = "different transaction collision"). note sure exactly what that means, and what the other transaction is, but the phy is never updated 2M.

[sniffer capture](https://github.com/zephyrproject-rtos/zephyr/files/7756478/different_transaction.zip)

note: this is another central (android phone with snapdragon 888) than the problematic macs above.

jsiverskog, 2021-12-22:
when i test between two zephyr devices (running the same code and phy update logic) everything seems to work. phy is updated to 2M as expected.

jsiverskog, 2021-12-22:
.. and if i set a timer for the android/snapdragon 888 case and trigger the phy update a tad later using a timer it seems to work as well.

thank you for your help @cvinayak

closed 2021-12-22 by jsiverskog

cvinayak, 2021-12-22:
@jsiverskog using a timer could be a bit flakky, and may expose the issue if control procedures are delayed. But sure, this issue can be closed. Feel free to open a new one if you come up with a solution that could be implemented as a quirk for phone out there already in the wild.