zephyrproject-rtos/zephyr #41375: hal_nordic: update nordic_hal to newest version

MeisterBob, 2021-12-21:
**Is your enhancement proposal related to a problem? Please describe.**
actually I'm strugling with an issue in the frame filter of nordic's IEEE802.15.4 driver (https://discord.com/channels/720317445772017664/883445320812466209/922478238155018280).

**Describe the solution you'd like**
The issue is already fixed in the driver included in NCS, please update zephyr's `hal_nordic` to the latest version.

**Describe alternatives you've considered**
I could create a pull request, that fixes the bug (just three lines of code), but I'm pretty sure there are more improvements in the latest version of the hal.

carlescufi, 2021-12-21:
@pkwiek @jciupis @e-rk @ankuns FYI, the community requests an update.

ankuns, 2021-12-21:
Please note PR with update of the nrf_802154:
https://github.com/zephyrproject-rtos/hal_nordic/pull/108
https://github.com/zephyrproject-rtos/zephyr/pull/41383

closed 2022-02-24 by carlescufi