zephyrproject-rtos/zephyr #41380: stm32h7/f7: Ethernet: Migrate driver to the new eth HAL api.

erwango, 2021-12-21:
**Is your enhancement proposal related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

erwango, 2021-12-21:
FYI: @Nukersson, @lochej (as code owners), @alsrgv(as ptp contributor).
I'm assigning the issue to myself for now (as if I don't do it, someone else will do anyway),
but please don't hesitate to take it, if interested.

alsrgv, 2021-12-23:
@erwango, I can help verify PTP functionality once you have a PR up.

closed 2022-12-05 by carlescufi