zephyrproject-rtos/zephyr #41399: samples: userspace: syscall_perf test cannot be run

gopiotr, 2021-12-22:
**Describe the bug**
From the beginning `sample.syscall_performances` test from `samples/userspace/syscall_perf` had wrong platform name in `platform_allow` list. This impact that this test was never performed in CI and it was "silently" skipped. I was trying to establish what kind of platform should be used for this test and I verify, that the only one RISCV platforms which fill the demanding for have `CONFIG_ARCH_HAS_USERSPACE` option enabled are `qemu_riscv32_xip` and `hifive1_revb`. I tried to build this sample for those platforms by `west` but it failed. 

**To Reproduce**
Steps to reproduce the behavior:
1. Run west command:
`west build -p always -b qemu_riscv32_xip samples/userspace/syscall_perf/`
or
`west build -p always -b hifive1_revb samples/userspace/syscall_perf/`
2. Observe the error during building.

**Expected behavior**
It should be possible to build this sample for some of RISCV platform, and run test dedicated for this sample.

**Impact**
Could not run `sample.syscall_performances` test in CI - it cause misleading information that such kind of test exists in Zephyr repository but in reality it never was run and tested.

**Logs and console output**
West logs from building this sample for `qemu_riscv_xip`:

> [1/155] Preparing syscall dependency handling
> 
> [18/155] Building C object CMakeFiles/app.dir/src/test_supervisor.c.obj
> /home/redbeard/zephyrproject/zephyr/samples/userspace/syscall_perf/src/test_supervisor.c: In function 'supervisor_thread_function':
> /home/redbeard/zephyrproject/zephyr/samples/userspace/syscall_perf/src/test_supervisor.c:28:3: warning: statement with no effect [-Wunused-value]
>    28 |   k_current_get();
>       |   ^~~~~~~~~~~~~~~
> [20/155] Building C object CMakeFiles/app.dir/src/test_user.c.obj
> /home/redbeard/zephyrproject/zephyr/samples/userspace/syscall_perf/src/test_user.c: In function 'user_thread_function':
> /home/redbeard/zephyrproject/zephyr/samples/userspace/syscall_perf/src/test_user.c:28:3: warning: statement with no effect [-Wunused-value]
>    28 |   k_current_get();
>       |   ^~~~~~~~~~~~~~~
> [133/155] Linking C executable zephyr/zephyr_pre0.elf
> FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/redbeard/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map 
> : && ccache /home/redbeard/zephyr-sdk-0.13.1/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/riscv/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/redbeard/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/riscv/core/pmp/libarch__riscv__core__pmp.a  zephyr/arch/arch/riscv/core/libarch__riscv__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/redbeard/zephyr-sdk-0.13.1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.3.0/rv32imac/ilp32"  -L/home/redbeard/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mabi=ilp32  -march=rv32imac  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /home/redbeard/zephyrproject/zephyr/build/zephyr && /usr/local/bin/cmake -E echo
> /home/redbeard/zephyr-sdk-0.13.1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.3.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/zephyr_pre0.elf section `datas' will not fit in region `RAM'
> /home/redbeard/zephyr-sdk-0.13.1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.3.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: region `RAM' overflowed by 28 bytes
> collect2: error: ld returned 1 exit status
> ninja: build stopped: subcommand failed.
> FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /home/redbeard/zephyrproject/zephyr/build

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK `zephyr-sdk-0.13.1`
 - Commit SHA used: 94a01d01924d1388ecaeab795c85cfe4fe59eda3

**Additional context**
After discussion with @nashif he proposed that this sample should be fixed and establish what kind of RISCV platform should be used to run it. Temporarily I proposed to skip this test until it will be fixed in this PR: #41377

enjiamai, 2021-12-22:
Hi @gopiotr , this https://github.com/zephyrproject-rtos/zephyr/pull/41416 might address the build issue by west. But I think there is still some problem need to be solved in this sample.

enjiamai, 2021-12-23:
I found the board configuration/setting "qemu_riscv32_xip" seems to be problematic. in this sample case,  it triggered an exception when we call k_sleep(), even in supervisor mode. The exception happens while CONFIG_USERSPACE=y. 

Reproduce step:
1. Apply #41416
2. Run  west build samples/userspace/syscall_perf/ -p always -b qemu_riscv32_xip -t run

You can see following error message in this sample case:

```
[155/156] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
             ROM:       61084 B        12 MB      0.49%
             RAM:       15628 B        16 KB     95.39%
        IDT_LIST:          0 GB         2 KB      0.00%
[155/156] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: riscv32
*** Booting Zephyr OS build v2.7.99-2681-g29dcd7ca9290  ***
Main Thread started; qemu_riscv32_xip
Supervisor thread started
E:
E:  mcause: 1, Instruction Access fault
E:   mtval: 20400be4
E:      a0: 20400d00    t0: 204028f0
E:      a1: 00000000    t1: 20400c04
E:      a2: 00000000    t2: 7fffffff
E:      a3: 00000000    t3: 00000000
E:      a4: 00000001    t4: 00000000
E:      a5: 80000000    t5: 00000000
E:      a6: 00000000    t6: 00000000
E:      a7: ffffffff
E:                      tp: 00000000
E:      ra: 204029c8
E:    mepc: 20400be4
E: mstatus: 00000080
E:
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80000118 (unknown)
E: Halting system
```
Hi, @cwshu  , I saw you contributed this board setting, could you please take a look at this?

cwshu, 2021-12-26:
@enjiamai 
After debugging, I found this error is the same bug as #39922.

It is an bug of un-aligned PMP region when using RISC-V userspace in XIP mode. RISC-V userspace configure `__rom_region_[start|size]` as the PMP region, but these symbols doesn't match power-of-2 alignment [1] in XIP mode.
(p.s. `MPU_ALIGN()` is only applied in the non-XIP linker script, and it is difficult to apply it in the XIP one.)

I have already fixed issue #39922 in the PR #40647. It is a small PR (2 commits) but depends on an unmerged large PR #39299 (12 commits). PR #39299 is ready but still lacks of review and approval from RISC-V reviewers.

[1] Power-of-2 alignment: size is power-of-2 && base is alignment to size. It is required for RISC-V NAPOT PMP region.

cwshu, 2021-12-26:
I have check `syscall_perf` on `qemu_riscv32_xip` works if applying your fix (#41416) on PR #40647.
(https://github.com/cwshu/zephyr/tree/syscall_perf)

p.s. The cycle/instr statistics is not accurate in QEMU because QEMU RISC-V doesn't emulate performance counter well. (At least I found cycle/instr counter is same value in QEMU RISC-V.)
I guess origin developer (@amergnat) run this sample on the real board. (e.g. hifive1_revb)

enjiamai, 2021-12-27:
Great! Thanks for your debugging and information! @cwshu  :) , looking at your PRs.

PR #41416 (2022-01-04): samples: userspace: fix syscall_perf test cannot be run
The syscall_perf test cannot be built successfully due to region RAM
overflow on qemu_riscv32_xip and hifive1_revb. The dtim is only 16KB,
To reduce the main stack size to make it can be built successfully.

Base on the spec([fe310-g002-manual-v1p1.pdf](https://sifive.cdn.prismic.io/sifive/e36c0a2f-413f-49d4-85c1-51ee1ed243e5_fe310-g002-manual-v1p1.pdf)), the size of itim should
be 8KB. Modify the definition in its DTS file.

And it also fix some of the gcc warning when testing it by twister.

This PR partially fixes #41399
commit 27db5646 (2022-01-04): samples: userspace: fix syscall_perf test cannot be run

The syscall_perf test cannot be built successfully due to region RAM
overflow on qemu_riscv32_xip and hifive1_revb. The dtim is only 16KB,
To reduce the main stack size to make it can be built successfully.

Base on the spec(fe310-g002-manual-v1p1.pdf), the size of itim should
be 8KB. Modify the definition in its DTS file.

And it also fix some of the gcc warning when testing it by twister.

This PR partially fixes #41399.
closed 2022-01-04 by nashif