zephyrproject-rtos/zephyr #41404: if zsock_connect() fails, tls_context does not get released automatically

nimarty, 2021-12-22:
**Describe the bug**
if zsock_connect() fails, tls_context does not get released. Subsequent calls to zsock_connect() always fail because tls_context cannot be allocated anymore. 

 - Platform: nrf91, mbedtls, zephyr sockets
 - Workaround: always call zsock_close() if a zsock_connect() fails to make sure tls_context gets released.

**To Reproduce**
Steps to reproduce the behavior:
1. build some firmware with zephyr networking stack and mbed tls which connects to a server in a loop
2. in the loop: allocate and connect socket (with TLS) to server
3. cut the connection to the server and watch the logoutput. there will be connection failures
4. make the server accessible again. subsequent connection attempts will fail even if they shouldn't

**Expected behavior**
if zsock_connect() fails I would expect any already allocated resources to be released so that the workaround of calling zsock_close() explicitely would not be necessary.

**Impact**
I invested quite some time to understand the problem and figure out a solution for this. Probably others will stumble upon this too.

**Logs and console output**
n/a

**Environment (please complete the following information):**
 - Zephyr v2.6.0

**Additional context**
n/a

rlubos, 2021-12-22:
I don't think calling `zsock_close` after connection failure is a workaround - it's rather an expected behavior. POSIX specification of the `connect()` function says (https://linux.die.net/man/3/connect):
> If connect() fails, the state of the socket is unspecified. Conforming applications should close the file descriptor and create a new socket before attempting to reconnect.

nimarty, 2021-12-23:
Hi @rlubos . Thank you for pointing that out. Fair enough if this is how the POSIX `connect()` works.

rlubos, 2022-01-03:
Closing based on the above.

closed 2022-01-03 by rlubos