zephyrproject-rtos/zephyr #41408: Low power states for H7

benediktibk, 2021-12-22:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
Currently there are no low power states implemented for the STM32 H7-series.

**Describe why you are asking for this support?**
I need this as a requirement for a project. I'm already working on the implementation of the power states, but I stumbled upon an issue with SYSTICK. On this platform by default SYSTICK is clocked by SYSCLK, which is stopped when the CPU enters CStop, which in turn is a requirement for the system to get into the state Stop. As the kernel sets up a timer based upon SYSTICK it will then never wake up again.
For testing purposes I already implemented a wakeup by RTC, which then of course completely confuses the kernel as SYSTICK did not continue in the meantime.

For comparison I took a look at the low power state implementation of STM32 L4 in zephyr. For this device the kernel also seems to assume that SYSTICK will continue. Unfortunately I am not absolutely sure, just based upon reading the reference manual of the L4, if SYSTICK will continue while the system is for instance in the mode Stop0.

My basic question is: Would it be okay to contribute a low power state implementation of H7 which won't wake up based upon the kernel scheduling anymore, only based upon an external wakeup source?
I assume not, which would then bring me to my next question: Is there a possibility to circumvent this issue? I saw for instance that SYSTICK could run from an external clock, but so far I could not figure out what this external clock would be.

erwango, 2021-12-22:
@benediktibk When using PM, Zephyr tick should be fed with lptim thanks to `drivers/timer/stm32_lptim_timer.c` driver (instead of cortex systick). This is how PM is working on L4/WB/L5/U5
So the idea is first to get LPTIM working on H7 (if not done yet).
Then enabling PM and lptim node will get things working (grep STM32_LPTIM_TIMER for more details).

benediktibk, 2021-12-23:
Started with implementation in #41439

benediktibk, 2021-12-23:
Maybe somebody can help met out a little bit, as I'm currently scratching my due to ARM usage fault which happens during the print of the boot banner. It happens only if the LPTIM is used as the system clock, but the debugger does not hit a breakpoint on lptim_irq_handler. Hence I am kinda confused where this is coming from. Actually no function in stm32_lptim_timer.c besides sys_clock_driver_init is getting called.

Would be glad if somebody could point me towards a useful direction, as I am running out of ideas how to investigate this further.

erwango, 2021-12-23:
> Would be glad if somebody could point me towards a useful direction, as I am running out of ideas how to investigate this further.

I would suggest testing H7 lptim driver implementation using `tests/kernel/timer/timer_api`, using `prj_tickless.conf`.
You'll probably miss some tuning to this driver to work on H7.

benediktibk, 2022-01-17:
Implemented by:

- https://github.com/zephyrproject-rtos/zephyr/pull/41871
- https://github.com/zephyrproject-rtos/zephyr/pull/41869
- https://github.com/zephyrproject-rtos/zephyr/pull/41867

closed 2022-03-21 by benediktibk