zephyrproject-rtos/zephyr #41422: The option CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE blocks k_sleep when CONFIG_PM is enabled

r2r0, 2021-12-22:
**Describe the bug**
If CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE and CONFIG_PM options are enabled together then k_sleep does not return.
According to documentation only uptime can be affected by enabling CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE option.

Platform is STM32WB55 Nucleo development board.
For blinky sample when CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE and CONFIG_PM options are set then LED is turn on all the time.
Without CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE option LED blinking is seen as expected.
Adding a few printk revealed that there is no code executed after call to k_msleep.

**To Reproduce**
Steps to reproduce the behavior:
Build simple blinky application with CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE and CONFIG_PM options enabled.

**Expected behavior**
The CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE should have no effect on sleep functionality..

**Impact**
annoyance

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.13.1
 - Commit SHA f92542ecf3e465861ae63fc05932fb229b527801

**Additional context**
Maybe it is a problem with following line in the next_timeout (in timeout.c) - it looks that `ret` will be always equal to -1 when CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE is set:
```C
	int32_t ret = to == NULL ? MAX_WAIT
		: CLAMP(to->dticks - ticks_elapsed, 0, MAX_WAIT);
```

r2r0, 2022-01-21:
@andyross Any chance to look at this - at first glance it looks as simple CLAMP parameter bug (upper bounding is less than lower bounding).

r2r0, 2022-01-25:
I would be happy to help fixing that but I have one doubt.
My understanding was that the `CLAMP` we talking about handles the case where there is requested a real timeout (system has something on the list waiting for timeout).
And then in this case a `CLAMP` is supposed only to limit timeout value to be in result type (`int32_t`) range.
Am I right? If not then where is the error in my reasoning?

andyross, 2022-01-25:
The logic there is supposed to be "the next timeout is the time in the future of the next item on the list, UNLESS it is negative[1] or longer than we can represent".  If all the inputs were positive, the CLAMP() macro would be exactly what we want. But that breaks because K_TICKS_FOREVER happens (for good reason) to have a negative value even though it's being used as MAX_WAIT (i.e. the **largest** possible value).

[1] Which can happen due to very late interrupts.

r2r0, 2022-01-26:
@andyross I am sorry I need to rise again topic about CLAMP role in the code under investigation.
I analysed again timeout.c and I really cannot find any place where `dticks` field can get value equal to `K_FOREVER` .
All assignments to `dticks` are always positive integers.
It means that in the following code we need CLAMP only to protect against late interrupts (`to->dticks - ticks_elapsed` expression has value lower than `0`).
```C
int32_t ret = to == NULL ? MAX_WAIT : CLAMP(to->dticks - ticks_elapsed, 0, INT32_MAX);
````
It means also that the only case when `next_timeout` will return `K_FOREVER`  is when `timeout_list` list is empty what is handled by the condition and true parts of the above conditional operator.

rljordan, 2022-02-08:
Removed @ceolin  as this is expected to be an issue with scheduler/timing.

rljordan, 2022-02-08:
Actually, @ceolin  will take the lead on this issue.

ceolin, 2022-02-08:
Ok, if I understood correctly we should still return `MAX_WAIT` in situations where the next timeout is longer than we can represent (`INT_MAX`). The problem with the current code is that when `CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE` is set, `MAX_WAIT` will be `K_TICKS_FOREVER` that is a negative number so `CLAMP` will end up always returning `MAX_WAIT`

```
CLAMP(val, low, high) (((val) <= (low)) ? (low) : MIN(val, high))
```

because in `MIN(val, high)`  the `high` value is a negative number. I think it can be solved in the following way:

```
	if ((to == NULL) ||
	    ((int64_t)(to->dticks - ticks_elapsed) > (int64_t)INT_MAX)) {
		ret = MAX_WAIT;
	} else {
		ret = MAX(0, to->dticks - ticks_elapsed);
	}
```

ceolin, 2022-02-13:
> Thanks @ceolin for picking it up. I am not sure for this: "...we should still return MAX_WAIT in situations where the next timeout is longer than we can represent (INT_MAX)..." IMO timer should be disabled **only** if there is no waiting nodes on timeout list. If system has at least one node on the timeout list then we should never disable system timer, even if the requested timeout is bigger than we can pass to system timer. In the latter case we can just "split" timeout to smaller chunks with values acceptable for system timer.

I tend to agree with it, but that is design decision IMHO. It seems that the deired behavior is that one described. I think that patch solves the problem of the system going to sleep with K_TICKS_FOREVER for any timeout when these options are enabled.
What do you think to open another issue / teature request to discuss this new proposed behavior ?

r2r0, 2022-02-14:
Nope, it is ok for me just a nit.

PR #42614 (2022-02-15): kernel: Fix timeout issue with SYSTEM_CLOCK_SLOPPY_IDLE
We can't simply use CLAMP to set the next timeout because
when CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE is set, MAX_WAIT is
a negative number and then CLAMP will be called with
the higher boundary lower the lower boundary.

Fixes #41422
commit 47b7c2e9 (2022-02-15): kernel: Fix timeout issue with SYSTEM_CLOCK_SLOPPY_IDLE

We can't simply use CLAMP to set the next timeout because
when CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE is set, MAX_WAIT is
a negative number and then CLAMP will be called with
the higher boundary lower the lower boundary.

Fixes #41422
commit 675a349e (2023-02-03): kernel: Fix timeout issue with SYSTEM_CLOCK_SLOPPY_IDLE

We can't simply use CLAMP to set the next timeout because
when CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE is set, MAX_WAIT is
a negative number and then CLAMP will be called with
the higher boundary lower the lower boundary.

Fixes #41422

(cherry picked from commit 47b7c2e93123d7100a97caea01910cb2e00af78b)