zephyrproject-rtos/zephyr #41449: PWM capture with STM32

bfa8, 2021-12-24:
**Describtion**
I work with the NUCLEO-H743ZI2, Zephyr SDK version 0.13.2 and Zephyr version 2.7.99 (build: v2.7.99-2467-g5cb3722a36d9). I needed to integrate the PWM capture to measure the period between two pulses. For this, I use the PWM peripherals from Zephyr  (https://docs.zephyrproject.org/latest/reference/peripherals/pwm.html). Concrete I used the pwm_pin_capture_nsec.

On this function, you can add the timeout (timeout – Waiting period for the capture to complete). In case this timeout elapses, on the STM32 PWM diver (pwm_stm32.c) an error log is generated (line 375 -> "No PWM capture callback specified"). The reason for it is, in the file pwm_capture.c on line 52 there the PWM first is configured properly, then it's started and on line 66 it's waiting for a semaphore or until the timeout elapse. If the time elapse the PWM is disabled (line 68) and after configured again, where the callback is NULL (I guess it is to unsubscribe the callback). But this call generate the error message in the STM32 PWM driver.

My question now, is the handling of the error in the STM32 PWM driver wrong or is there a bug in the pwm_capture.c by configure the PWM without a valid callback?

**Expected behavior**
I expect, that no error is generated if the timeout of the PWM capture is elapsed.

**Impact**
This bug is an annoyance.

**Logs and console output**
[00:56:33.418,000] <err> pwm_stm32: No PWM capture callback specified           
[00:56:33.418,000] <wrn> pwm_capture: pwm capture timed out    

**Environment (please complete the following information):**
 - Linux
 - Zephyr SDK Version 0.13.2
 - Zephyr Version 2.7.99

**Additional context**
I used for the input capture the PA0 with the timer 2 on channel 1.

ABOSTM, 2022-01-05:
@bfa8, you are right this is abnormal.
And looking at other implementations:
* https://github.com/zephyrproject-rtos/zephyr/blob/c51aa88046ca5c9d687e3e6802b32ddd57dd6de4/drivers/pwm/pwm_mcux_ftm.c#L138
* https://github.com/zephyrproject-rtos/zephyr/blob/c51aa88046ca5c9d687e3e6802b32ddd57dd6de4/drivers/pwm/pwm_mcux_pwt.c#L66

there is no check on CallBack parameter.
So STM32 driver is wrong and should not check on `cb `parameter.
(please note that STM32 capture implementation is quite fresh : 2 month old)

Do you want to submit a Pull Request to fix this issue ?

bfa8, 2022-01-05:
> @bfa8, you are right this is abnormal. And looking at other implementations:
>
> * [...]

Yes, I would like to submit a Pull Request. I will add the link to the PR to this report as soon as I have created it.

erwango, 2022-01-11:
@bfa8 Do you still plan to push a fix for this ?

bfa8, 2022-01-11:
@erwango Yes, I still plan to do it. I took a look last week and at first thought it was low-hanging fruit and tried it out. But unfortunately, I ran into a problem with just deleting the callback check. Since I was running out of time, I postponed this task to really get to the bottom of the issue.

FRASTM, 2022-03-17:
Is it just a question of removing 
```
-	if (cb == NULL) {
-		LOG_ERR("No PWM capture callback specified");
-		return -EINVAL;
-	}
```

in the pwm_stm32_pin_configure_capture() 
and maybe also cpt->callback if not NULL in the pwm_stm32_isr()

closed 2022-03-17 by MaureenHelm