zephyrproject-rtos/zephyr #41465: Periodic advertising sync failure, when "DONT_SYNC_AOA" or "DONT_SYNC_AOD" options is used

saleh-unikie, 2021-12-27:
**Describe the bug**
I have developed a sample application based on ["direction_finding_connectionless_rx" sample](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/direction_finding_connectionless_rx). This new sample supports receiving multiple signals from different tags/beacons in parallel, which is needed to catch the bug I am going to describe it here.
You can find this new sample here: https://github.com/saleh-unikie/direction_finding_connectionless_rx_multiple

Every thing is normal until when one of these options has been added to the code:
```
BT_LE_PER_ADV_SYNC_OPT_DONT_SYNC_AOA
BT_LE_PER_ADV_SYNC_OPT_DONT_SYNC_AOD_1US
BT_LE_PER_ADV_SYNC_OPT_DONT_SYNC_AOD_2US
```

For example by adding `BT_LE_PER_ADV_SYNC_OPT_DONT_SYNC_AOA` to the above example, it is expected that the receiver does NOT make any synch with the AoA tag.
```
...
sync_create_param.options = BT_LE_PER_ADV_SYNC_OPT_SYNC_ONLY_CONST_TONE_EXT |
			    BT_LE_PER_ADV_SYNC_OPT_DONT_SYNC_AOA;
...
```
It works fine until when only AoA tags or only AoD beacons are in the environment, If both types (AoA tag + AoD beacon) are in the environment, then it makes synch with both types, while it must not make any sync with AoA tags (according to the above options).
Also sometimes it generates "assertion failure" as it has shown in the "Logs and console output"

**To Reproduce**
Steps to reproduce the behavior:
1. cd _project_
2. west build -b nrf52833dk_nrf52833
3. west flash
4. Run at least one AoA tag and one AoD beacon, based on ["direction finding connectionless tx"](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/direction_finding_connectionless_tx) 
5. Connect a terminal to the receiver
6. See error (doesn't happen every time, needed to be restarted periodically)

**Expected behavior**
The receiver must reject making sync with AoA tags and only sync with AoD beacons (dependent on the used options)

**Impact**
It fails working normally when both AoA tags and AoD beacons are available.

**Logs and console output**
Different errors are seen and it doesn't happen every time. so it is needed to restart the receiver until when a failure has been seen. Restarting the tags/beacons while receiver is working, can also affect on this behavior. These two errors are seen in different rounds:
```
ASSERTION FAIL [(ticker_status == 0) || (ticker_status == 2)] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_scan_aux.c:531
```
```
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull.c:1216
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain :Zephyr SDK
 - Commit SHA or Version used: commit 770732fe92faab44d93bcbc00fc4d45476c24dea
 - HW: nRF52833-DK

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

cvinayak, 2021-12-28:
@saleh-unikie I am in the process of changing the periodic advertising implementation for various missing race and memory release scenarios. Could you please use the work in progress changes to the controller in this PR: https://github.com/zephyrproject-rtos/zephyr/pull/41091

Do revert back if the report issues still exist when using changes in #41091

saleh-unikie, 2021-12-28:
Thanks a lot @cvinayak 
I've tested the PR and above crash hasn't seen any more, but another problem still exist.

The scenario to generate the problem:
Using rx source code (using above code that I've provided to work with simultaneous tx nodes) and setting the `options` variable of `struct bt_le_per_adv_sync_param` in order NOT to sync with AoA tags:
```
struct bt_le_per_adv_sync_param sync_create_param;
...
sync_create_param.options = BT_LE_PER_ADV_SYNC_OPT_DONT_SYNC_AOA;
...
bt_le_per_adv_sync_create(&sync_create_param, synch);
```

**Normal operation**:
If only AoD beacons are available, then it works fine. (I've tested with two AoD beacon)

**Problematic operation**:
If one AoD beacon and one AoA tag are available, then after making the first sync with AoD beacon (but it doesn't make any sync with AoA tag properly), it doesn't make another sync with AoD beacon again if it has been restarted.
In fact it doesn't report any AoD beacon at `scan_recv` callback function after a while, even if scan has been disabled and enabled again.

I noticed availability of an AoA tag, can prevent the rx side to work normally if `BT_LE_PER_ADV_SYNC_OPT_DONT_SYNC_AOA` has been used, despite of the sequence we start the devices.
For example if at the beginning only AoD beacon is available, then it works fine until when an AoA tag starts. Or if at the beginning only an AoA tag is available and after a while an AoD beacon has turned on, then it doesn't sync with the AoD tag at all. So in summary:

**Summary**
Scenario A:
Turn on rx (DONT_SYNC_AOA) --> Turn on AoD beacon (every thing works fine) --> Turn on AoA tag (after a while `scan_recv` is not called anymore)
Scenario B:
Turn on rx (DONT_SYNC_AOA) --> Turn on AoA tag (after a while `scan_recv` is not called anymore) --> Turn on AoD beacon (`scan_recv` is not called, so no sync has created)

saleh-unikie, 2021-12-28:
@cvinayak Regarding the problem I've described in the last comment, I've suspected to this part of the code:
file: subsys/bluetooth/controller/ll_sw/nordic/lll/lll_sync.c
```
static void isr_rx_adv_sync_estab(void *param)
{
....
    if (sync_ok == SYNC_STAT_ALLOWED) {
        //works fine here
    } else if (sync_ok == SYNC_STAT_TERM) {
        //whenever enters here, then problem has seen
    }
...
}

```

Every time `lll_sync_cte_is_allowed` function returns with `SYNC_STAT_TERM`, then not only it prevents the system to make a sync (expected behavior) but also stops scanning and `scan_recv` function will not be called at the application layer anymore.

PS: I forced `lll_sync_cte_is_allowed` to return `SYNC_STAT_READY_OR_CONT_SCAN` and after that, not only a sync wasn't created (expected) but also `lll_sync_cte_is_allowed` was called repeatedly after that seems it isn't normal

carlescufi, 2022-01-12:
@ppryga will you please take a look and post a fix if applicable?

ppryga, 2022-02-14:
Sorry for long time the issue was waiting.

@saleh-unikie I was not able to reproduce the issue with use of 3DKs. One was working with `direction_finding_connectionless_rx`, two other were `direction_finding_connectionless_tx` where one was sending AoA and another AoD.
I've set: `BT_LE_PER_ADV_SYNC_OPT_SYNC_ONLY_CONST_TONE_EXT | BT_LE_PER_ADV_SYNC_OPT_DONT_SYNC_AOD_2US` as a `sync_create_param.options`.

Filtering by CTE type was working correctly. I didn't found an issue in Controller or Host code.

I've found a bug in `direction_finding_connectionless_rx` sample. Here is a PR: https://github.com/zephyrproject-rtos/zephyr/pull/42792 (I'll create an issue for that also).

Can you try once again if the issue exist? Can you take a look on PR and see if there is something that may also impact your test code?

saleh-unikie, 2022-02-15:
Thanks @ppryga for your time.

Yes, seems the issue is fixed using the last upstream/main commit. When it was reported, the https://github.com/zephyrproject-rtos/zephyr/commit/770732fe92faab44d93bcbc00fc4d45476c24dea commit was used which was not working correctly.
Only maybe it is needed to update some comments at `bluetooth.h` by adding `Don't` at the beginning of the comment:
https://github.com/zephyrproject-rtos/zephyr/blob/main/include/bluetooth/bluetooth.h#L1285

And also thanks a lot for the PR #42792 , I reflected these changes to my code (simultaneous tag support) too. It enhanced the performance.

saleh-unikie, 2022-02-15:
UPDATE
By running the sample code for a longer time, I've detected a problem but I'm not sure it is related to this issue or not.

**Test Scenario**: 4 AoA-Tag are synced with AoA-Locator. Then an AoD-Beacon start advertising and repeatedly tries to make sync with the locator, but fails at each iteration (expected behavior), but after a while the locator stops receiving the sync requests from AoD-Beacon. At this time, I turned on two another AoA-Tag but AoA-Locator didn't respond to them.

ppryga, 2022-02-17:
@saleh-unikie  I've found root cause of behavior you have observed. The fix is here: https://github.com/zephyrproject-rtos/zephyr/pull/42922. Could you confirm if that works with your setup. I've checked it using samples provided by you but I'd like to have it double checked. Thanks in advance!

PR #42922 (2022-02-17): Bluetooth: Controller: df: Fix missing node_rx release if wrong CTE type
Periodic advertising sync may be created with CTE type filtering
enabled. When received AUX_SYNC_IND, there are HCI_LE_Periodic_-
Advertising_Sync_Established and HCI_LE_Periodic_Advertising_Report
events generated. In case received AUX_SYNC_IND has wrong CTE type,
HCI_LE_Periodic_Advertising_Sync_Established event is generated.
There may not be HCI_LE_Periodic_Advertising_Report event generated.
In this case node_rx, prepared by LLL, was not released in ULL.
It was lost in ull_sync_established_report.

Fixes #41465.
closed 2022-02-17 by carlescufi