zephyrproject-rtos/zephyr #41469: Proposal to integrate Synopsys SDMMC driver from Trusted Firmware-A

ngboonkhai, 2021-12-28:
## Origin
Name of Project : Trusted Firmware-A

Project Link: https://github.com/ARM-software/arm-trusted-firmware

Source Code:
https://github.com/ARM-software/arm-trusted-firmware/blob/master/drivers/mmc/mmc.c
https://github.com/ARM-software/arm-trusted-firmware/blob/master/drivers/synopsys/emmc/dw_mmc.c
https://github.com/ARM-software/arm-trusted-firmware/blob/master/include/drivers/synopsys/dw_mmc.h
https://github.com/ARM-software/arm-trusted-firmware/blob/master/include/drivers/mmc.h

## Purpose

The dw_mmc.c driver provide the access to Sypnopsys Designware MMC controller to perform read and write operation
While the mmc.c is a general interface how to communicate with a general MMC Controller.

## Mode of integration

These codes would intergrate to the zephyr main tree.

## Pull Request
https://github.com/zephyrproject-rtos/zephyr/pull/41468

## Description

It is a Synopsys SDMMC diver which help to access the Synopsys SDMMC controller such and read and write operation.

The SDMMC driver is easy to port from ARM trusted firmware, and can be reuse in zephyr wihout much modification,

It is a Opensource BSD-3-Clause license.

The Intel SoC FPGA Agilex family device is using the Synopsys SDMMC controller and we are trying to enable the SDMMC driver
in zephyr so that the Intel SoC FPGA Agilex device can perform SDMMC operation using zephyr. Integrate to zephyr will help for enablement of
future device also, anyone are free to use the SDMMC driver from zephyr.

# Dependencies
No.

It will be included as an abstraction layer with this component and it is a
replaceable implementation.

## Revision
770732fe92faab44d93bcbc00fc4d45476c24dea

## License
SPDX-License-Identifier: BSD-3-Clause

nashif, 2022-01-06:
proposal was brought up in the TSC and was approved, per guidelines, this will now be forwarded to the board.

nashif, 2022-01-18:
This was approved, PR can go.

ngboonkhai, 2022-01-19:
> This was approved, PR can go.

That's great, thanks Anas :D

henrikbrixandersen, 2023-08-31:
#41468 was closed - any progress on this?

nashif, 2024-02-08:
closing, please reopen if this comes back.

closed 2024-02-08 by nashif