zephyrproject-rtos/zephyr #41472: Unable to mount fat file system on nucleo_f429zi

maxstreitberger, 2021-12-28:
Hello everyone,
I am trying to set up a file system in the internal flash of the nucleo_f429zi. For that, I followed the [USB Mass Storage Sample Application](https://docs.zephyrproject.org/latest/samples/subsys/usb/mass/README.html) example and #33252. However, I receive this output:
```
Area 1 at 0x10000 on FLASH_CTRL for 65536 bytes
[00:00:00.005,000] <err> usb_msc: Storage init ERROR !!!! - Aborting USB init
[00:00:00.009,000] <err> fs: fs mount error (-5)
[00:00:00.009,000] <err> main: Failed to mount filesystem
[00:00:00.010,000] <inf> main: The device is put in USB mass storage mode.
```
I configured my project as following:
```
CONFIG_STDOUT_CONSOLE=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_PRODUCT="Zephyr MSC sample"
CONFIG_LOG=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_MASS_STORAGE=y
CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
CONFIG_USB_MASS_STORAGE_LOG_LEVEL_ERR=y
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_BOOTLOADER_MCUBOOT=y
CONFIG_MCUBOOT_SIGNATURE_KEY_FILE="bootloader/mcuboot/root-rsa-2048.pem"
CONFIG_FILE_SYSTEM=y
CONFIG_FAT_FILESYSTEM_ELM=y
CONFIG_DISK_DRIVERS=y
CONFIG_DISK_DRIVER_FLASH=y
CONFIG_DISK_FLASH_VOLUME_NAME="NAND"
CONFIG_DISK_FLASH_DEV_NAME="NUCLEO_FLASH"
CONFIG_DISK_FLASH_MAX_RW_SIZE=256
CONFIG_DISK_FLASH_ERASE_ALIGNMENT=0x1000
CONFIG_DISK_ERASE_BLOCK_SIZE=0x1000
CONFIG_DISK_FLASH_START=0x10000
CONFIG_DISK_VOLUME_SIZE=0x10000
CONFIG_DISK_FLASH_SECTOR_SIZE=512
CONFIG_FS_FATFS_MOUNT_MKFS=y
CONFIG_FS_FATFS_READ_ONLY=n
```
The flash configuration is based on this devicetree:
https://github.com/zephyrproject-rtos/zephyr/blob/770732fe92faab44d93bcbc00fc4d45476c24dea/boards/arm/nucleo_f429zi/nucleo_f429zi.dts#L166-L208

When I'm using RAM instead of flash, everything works as expected. Thus, I think I either forgot something or I misconfigured something.

**Expected behavior**
```
Mount /NAND:: 0
/NAND:: bsize = 512 ; frsize = 1024 ; blocks = 2028 ; bfree = 1901
/NAND: opendir: 0
  F 0 SAMPLE.TXT
End of files
[00:00:00.077,423] <inf> main: The device is put in USB mass storage mode.
```

erwango, 2022-01-03:
There's a known compatibility issue between flash based file system and F4 based targets, due to complex flah layout on these devices.
Cf https://github.com/zephyrproject-rtos/zephyr/issues/19895 and https://github.com/zephyrproject-rtos/zephyr/issues/28309.

These are open for long time and I'm not sure there's and easy solution that can be fitted.
Can you have a look to the link above and see if helping ?

maxstreitberger, 2022-01-25:
Ok, so I found out that I have to change `CONFIG_DISK_FLASH_DEV_NAME="NUCLEO_FLASH"` to `CONFIG_DISK_FLASH_DEV_NAME="FLASH_CTRL"` to have access to the flash controller. It then checks if there is a FAT volume on the drive. If not, it creates a new partition. After creating the partition, it checks if there is a FAT VBR (`zephyrproject/modules/fs/fatfs/ff.c:3247`), which fails. So I guess that my flash configuration is wrong, but I don't know what I have to change. I also looked at #33252 and tried it with this configuration, but it still failed.

erwango, 2022-01-25:
> Ok, so I found out that I have to change `CONFIG_DISK_FLASH_DEV_NAME="NUCLEO_FLASH"` to `CONFIG_DISK_FLASH_DEV_NAME="FLASH_CTRL"` to have access to the flash controller. It then checks if there is a FAT volume on the drive. If not, it creates a new partition. 

Thanks for the heads up

> After creating the partition, it checks if there is a FAT VBR (`zephyrproject/modules/fs/fatfs/ff.c:3247`), which fails. So I guess that my flash configuration is wrong, but I don't know what I have to change. I also looked at #33252 and tried it with this configuration, but it still failed.

Have you checked #19895 and #28309 as mentioned earlier ?
Also, maybe you can provide your flash config so we can have a look.

maxstreitberger, 2022-01-25:
My flash config is mentioned above. 
I looked at them, but I don't know how they help me.

de-nordic, 2022-02-08:
I do not think that FAT, with write support, can work on the STM32F429, at leas twit your config. I have looked at the flash driver for the device: https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/flash_stm32f4x.c#L263-L274
and this is layout of flash
```
#elif FLASH_SECTOR_TOTAL == 24
static const struct flash_pages_layout stm32f4_flash_layout[] = {
	/*
	 * RM0090, table 6: STM32F427xx, STM32F437xx, STM32F429xx, STM32F439xx
	 * RM0386, table 4: STM32F469xx, STM32F479xx
	 */
	{.pages_count = 4, .pages_size = KB(16)},
	{.pages_count = 1, .pages_size = KB(64)},
	{.pages_count = 7, .pages_size = KB(128)},
	{.pages_count = 4, .pages_size = KB(16)},
	{.pages_count = 1, .pages_size = KB(64)},
	{.pages_count = 7, .pages_size = KB(128)},
};
```
so the `CONFIG_DISK_ERASE_BLOCK_SIZE` and `CONFIG_DISK_ERASE_BLOCK_ALIGNMENT` have to reflect one of these, depending on where your FAT starts; the problem is that erase operation takes out entire page, so in the device case it would erase 16k, 64k or 128k at once.
You could probably still use read-only FAT, but you would have to program the partition with your application hex.

maxstreitberger, 2022-02-10:
@de-nordic, so you are saying it is not possible to have a read and write file system on the STM32F429 flash? Or it is just not possible with FAT? If it is the FAT, what would you recommend as a file system?

de-nordic, 2022-02-10:
> @de-nordic, so you are saying it is not possible to have a read and write file system on the STM32F429 flash? Or it is just not possible with FAT? If it is the FAT, what would you recommend as a file system?

I do not think it is possible to have any kind of true file system work with the memory layout.
Look at FAT: it uses 512B blocks, which means that when you have 4096B page  and need to overwrite one block, then you have to pick entire page to RAM, because there are four blocks in it, erase the target page and write one modified and seven other non-modified blocks back.
Now if you have 64k large erase page that means that there are 128blocks on it and overwrite of any block requires erasing all of them and writing some of them back.
File systems need to be able to allocate and be able to modify i-nodes and other structure they use for file system management and it is really hard to do when you have so few but large blocks, because basically erase page is a block - FAT simulates 512 blocks by division of page, as you may specify erase page and size of block independently.
I do not think that there is a file system you could use here with write support.

closed 2022-04-26 by github-actions[bot]