zephyrproject-rtos/zephyr #41482: kernel: Dummy thread should not have an uninitialized resource pool

berendo, 2021-12-29:
**Describe the bug**
The [dummy thread](https://github.com/zephyrproject-rtos/zephyr/blob/770732fe92faab44d93bcbc00fc4d45476c24dea/kernel/init.c#L414-L416)'s `resource_pool` is [inherited](https://github.com/zephyrproject-rtos/zephyr/blob/770732fe92faab44d93bcbc00fc4d45476c24dea/kernel/thread.c#L602) by other threads created during system initialization. Given that the current [`z_dummy_thread_init()`](https://github.com/zephyrproject-rtos/zephyr/blob/770732fe92faab44d93bcbc00fc4d45476c24dea/kernel/include/kswap.h#L214-L230) implementation does _not_ initialize the `resource_pool` element, this uninitialized value propagate to other threads, which may unknowingly "use" it if the uninitialized value ends up being non-`NULL`.

**To Reproduce**
Build Zephyr with multi-threading enabled, breakpoint at `main()` and inspect the `resource_pool`'s of various threads in the system:
```
...
Breakpoint 1, main () at /home/berend/scorpio-fw-4/src/scpu/src/main.c:59
59      {
(gdb) print z_main_thread->resource_pool
$1 = (struct k_heap *) 0xbf
(gdb) print z_idle_threads[0].resource_pool
$2 = (struct k_heap *) 0xbf
(gdb) print shell_uart_thread.resource_pool
$3 = (struct k_heap *) 0xbf
(gdb) print k_sys_work_q->thread.resource_pool
$4 = (struct k_heap *) 0xbf
(gdb) 
```
**Expected behavior**
An uninitialized `resource_pool` should not propagate to any threads. It should at the very least be set to `NULL` or, preferably, if `CONFIG_HEAP_MEM_POOL_SIZE > 0`, the system heap should be inherited via the dummy thread's `resource_pool`.

**Impact**
Unexplained crashes, memory corruption, etc. due to uninitialized pointer dereferencing when `_current->resource_pool` is used in various ways, including but not limited to `libc-hooks.c`'s dynamic kernel object allocation.

**Logs and console output**
N/A

**Environment (please complete the following information):**
The `v2.7.0` tag was used, but the issue may be endemic since the introduction of the `resource_pool` element in the `k_thread` structure back in 92e5bd7.

**Additional context**
I can proide a PR to address this issue in `z_dummy_thread_init()` with the inclusion of these lines:
```
#if (CONFIG_HEAP_MEM_POOL_SIZE > 0)
	k_thread_system_pool_assign(dummy_thread);
#else
	dummy_thread->resource_pool = NULL;
#endif
```

dleach02, 2022-01-04:
okay, after seeing the issue I agree with the change request being proposed. Please provide a PR.

I was going to say that the dummy thread is a temporary thread to facilitate the initialization flow as the forced context switch handling code expects a "thread" to be active on the switch to store the state information.. then the dummy thread isn't used anymore. But seeing what you wrote up it and the proposed PR it is a reasonable fix.

Thank you for the bug report and the offer to contribute a PR.

berendo, 2022-01-04:
> Please provide a PR.

Will do - thank you for the feedback @dleach02 and @dkalowsk.

berendo, 2022-01-05:
I opened the PR @dleach02.

PR #41578 (2022-01-06): kernel/swap: Initialize dummy thread's resource pool
This PR addresses the issue related to the dummy thread's uninitialized resource pool described in #41482.

Fixes #41482
commit b208e581 (2022-01-06): kernel/swap: Initialize dummy thread's resource pool

The resource pool of the short-lived dummy thread "stub" may be
inherited by other threads created during system initialization. This
commit initializes this resource pool to NULL or the system pool to
ensure that a well-defined resource pool propagates to other threads
that inherit it from the dummy thread.

Fixes #41482.
closed 2022-01-06 by nashif