zephyrproject-rtos/zephyr #41484: add comments to show examples of adding each feature in device binding file, like st,stm32-usart.yaml

wxzhuhua, 2021-12-29:
**Is your enhancement proposal related to a problem? Please describe.**
I was trying to move my code from zephye-2.3 to 2.7.99, got build errors about pinctrl-name and pinctrl-0, want to add them in dtsi file but don't know how.

**Describe the solution you'd like**
for example, in st,stm32-usart.yaml, add comments to show examples of adding each feature

**Describe alternatives you've considered**
give some description in doc or provide a dtsi file to show how for each kind of chip.

**Additional context**
none

wxzhuhua, 2021-12-30:
I found an example:
pinctrl-0 = <&usart1_tx_pa9 &usart1_rx_pa10>;
pinctrl-names = "default";

though, the builder complains: undefined node label 'usart1_tx_pa9', can't find a define for it

erwango, 2022-01-03:
@wxzhuhua We need to think about how to document that properly through bindings indeed.
This being said, for your specific case, node labels are available in `zephyrproject/modules/hal/stm32/dts`

wxzhuhua, 2022-01-04:
@erwango Sounds good, anywhere is fine, but there have to be somewhere.