zephyrproject-rtos/zephyr #41486: Zephyr project installation

srikanth359, 2021-12-29:
initialisation and updation of west tool in windows

After initialisation of west by using west init zephyrproject command then i gave the west update command then it updated for sometime and it is showing ERROR OCCURED UPDATE FAILED.

Expected behaviour is it should update properly and i followed all the steps properly from beginning.

OS : Windows 

ERROR SCREEN 

=== updating zscilib (modules/lib/zscilib):
HEAD is now at 12bfe3f Merge pull request #28 from microbuilder/doccleanup
ERROR: update failed for project trusted-firmware-m

Thanks in advance

Regards 
SRIKANTH P

ycsin, 2021-12-30:
It would be easier to assess what went wrong if you attach the full history of your terminal

srikanth359, 2021-12-30:
Thank you ycsin i solved that problem ,

but there is another bug it would be easy for me if you help me with this 

while building using west tool the cmd screen is showing error in windows OS

-- west build: generating a build system
Including boilerplate (Zephyr base): C:/Users/vimel/SRIKANTH/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/Users/vimel/SRIKANTH/zephyrproject/zephyr/samples/hello_world
-- Zephyr version: 2.7.99 (C:/Users/vimel/SRIKANTH/zephyrproject/zephyr), build: v2.7.99-2735-g8660457ec2cd
-- Found Python3: C:/Users/vimel/AppData/Local/Programs/Python/Python310/python.exe (found suitable version "3.10.1", minimum required is "3.6") found components: Interpreter
CMake Error at C:/Users/vimel/SRIKANTH/zephyrproject/zephyr/cmake/west.cmake:48 (message):
  Unable to import west.version from
  'C:/Users/vimel/AppData/Local/Programs/Python/Python310/python.exe':

  Traceback (most recent call last):

    File "<string>", line 1, in <module>

  ModuleNotFoundError: No module named 'west'

  Please install with:

      C:/Users/vimel/AppData/Local/Programs/Python/Python310/python.exe -m pip install west

  Or verify these installations:

    The Python version used by west is:  C:\Users\vimel\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe (real path C:/Users/vimel/AppData/Local/Microsoft/WindowsApps/PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0/python.exe)
    The Python version used by CMake is: C:/Users/vimel/AppData/Local/Programs/Python/Python310/python.exe
Call Stack (most recent call first):
  C:/Users/vimel/SRIKANTH/zephyrproject/zephyr/cmake/app/boilerplate.cmake:150 (include)
  C:/Users/vimel/SRIKANTH/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  C:/Users/vimel/SRIKANTH/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' '-DWEST_PYTHON=C:\Users\vimel\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe' '-BC:\Users\vimel\SRIKANTH\zephyrproject\zephyr\build' '-SC:\Users\vimel\SRIKANTH\zephyrproject\zephyr\samples\hello_world' -GNinja -DBOARD=esp32c3_devkitm